meshtastic/firmware #1395: [Bug]: Channel Index outside of scope should not freeze firmware

caveman99, 2022-04-22:
### Category

Other

### Hardware

Rak4631

### Firmware Version

1.3.8.2c04a06c-d

### Description

Attempting to decode a protobuf with the wrong key may lead to a situation where the assert from the log is triggered. if that happens, fail gracefully instead of halting execution.

### Relevant log output

```Shell
14:02:07 3381 [Router] Error: can't decode protobuf reason='invalid wire_type', pb_msgdesc 0x0x66bc0
14:02:07 3381 [Router] Invalid protobufs in received mesh packet (bad psk?)!
14:02:07 3381 [Router] No suitable channel found for decoding, hash was 0x8!
14:02:07 3381 [Router] packet decoding failed (no PSK?) (id=0x30999ca8 Fr0xf0 To0xff, WantAck0, HopLim3 Ch0x8 encrypted rxtime=1650636125 rxSNR=3.75 rxSNR=1.52895e-154)
14:02:07 3381 [Router] No one responded, send a nak
14:02:07 3381 [Router] Alloc an err=8,to=0xc4825df0,idFrom=0x30999ca8,id=0x4c53681a
14:02:07 3381 [Router] Add packet record (id=0x4c53681a Fr0x03 To0xf0, WantAck0, HopLim0 Ch0x8 Portnum=5 requestId=30999ca8 rxtime=1650636127 priority=120)
14:02:07 3381 [Router] assert failed src\mesh\Channels.cpp: 185, Channel& Channels::getByIndex(ChannelIndex), test=chIndex < channelFile.channels_count
```

caveman99, 2022-04-23:
There are other parts of the code that use Asserts. As a patch, we should add an NRF restart in __assert_func of main-nrf52.cpp, per @thebentern 

Then go through the asserts one by one and try to handle them gracefully. we use this meta-issue to track progress,

thebentern, 2022-04-26:
Related to #1407

caveman99, 2022-04-27:
Woke up to another logged assert today. Last line logged was:
`09:44:22 30054 [RadioIf] assert failed src\mesh\SX126xInterface.cpp: 164, void SX126xInterface<T>::setStandby() [with T = Exception in thread rx:`

sachaw, 2022-05-12:
May be worth updating radiolib first then seeing if this persists.

caveman99, 2022-08-10:
Radiolib was updated some time ago and is working well, if we find new asserts, reopen this report.

closed 2022-08-10 by caveman99