meshtastic/firmware #1396: [Bug]: Meshtastic 1.3.x

D4rk4, 2022-04-22:
### Category

Other

### Hardware

T-Lora v1

### Firmware Version

1.3.8.90df7c2

### Description

On TLORA_V1 not shown neighbors over CLI, but I can see incoming packets via debug.
I'm not sure it's firmware or meshtastic-cli related.

### Relevant log output

```Shell
dark@hope{~/automation}:meshtastic --info
Connected to radio

Owner: FRIEND (FRI)
My info: { "myNodeNum": 4066809116, "firmwareVersion": "1.3.8.90df7c2", "rebootCount": 13, "bitrate": 136.44214, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true }
Nodes in mesh:  {'num': 4066809116, 'user': {'id': '!f266951c', 'longName': 'FRIEND', 'shortName': 'FRI', 'macaddr': '08:3a:f2:66:95:1c', 'hwModel': 'TLORA_V1'}, 'position': {}}

Preferences: {}

Channels:

Primary channel URL: https://www.meshtastic.org/d/#

dark@hope{~/automation}:meshtastic --nodes
Connected to radio
╒═════╤════════╤═══════╤═══════════╤═══════╤═════════════╤═════════╕
│   N │ User   │ AKA   │ ID        │ SNR   │ LastHeard   │ Since   │
╞═════╪════════╪═══════╪═══════════╪═══════╪═════════════╪═════════╡
│   1 │ FRIEND │ FRI   │ !f266951c │ N/A   │ N/A         │ N/A     │
╘═════╧════════╧═══════╧═══════════╧═══════╧═════════════╧═════════╛
```

D4rk4, 2022-04-24:
As I discovered `meshtastic --set position_broadcast_secs` will reset the meshtastic region (and maybe another device settings).

thebentern, 2022-04-24:
@mc-hamster @caveman99, I think we should prioritize hunting this one down. I noticed the same issue yesterday afternoon when I was trying to set some telemetry prefs yesterday. I think we may have recently introduced a regression, because I recall setting multiple settings working fine on 1.3 in the past.

rwcts, 2022-04-28:
I have a bug opened for this on https://github.com/meshtastic/Meshtastic-python/issues/319 and just tested out the  1.3.10 and the same issues. It reset region but i can test further and find that it will reset all channel info.

garthvh, 2022-06-20:
This is fixed in 1.3.20

Both --nodes and --info are working, there is data missing.

you can also run 

--shutdown
--reset

And save region.

closed 2022-06-20 by garthvh