meshtastic/firmware #1406: [Feature Request]:

cnf, 2022-04-25:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

v1.2.59

### Description

Hi,

I am mounting a ttgo Lora in [this case](https://www.printables.com/model/131389-ttgo-lora32-v2116-case-for-meshtastic) with the DIN mount.

Because of the position of the antenna, mounted the screen is upside down for reading.

Would it be possible to add a config option to flip the screen 180 degrees, so it is presented better in those situations?

Relevant [discord talk here](https://discord.com/channels/867578229534359593/920180403799728210/967808852001173555)

Thanks!

caveman99, 2022-04-25:
This will be considered for 1.3 pending the protobuf config reshuffle.

cnf, 2022-04-25:
Thanks a lot! All I can ask for 😄

sachaw, 2022-09-11:
Protobufs implemented

closed 2022-10-04 by caveman99