meshtastic/firmware #1416: [Feature Request]: Long Range FHSS supporting

boby1975, 2022-04-27:
### Category

Other

### Hardware

T-Echo

### Firmware Version

1.2.59

### Description

Hi
I did not find **"Long Range FHSS settings"** in **[Protobuf API](https://meshtastic.org/docs/developers/protobufs/api#channelsettings)** documentation. 
Such technology is suppurted in Semtech modules **SX1261/2**. It would be nice to add this if possible. 
Thanks.

**PS.**
Tell me how to set it up if it already exists

caveman99, 2022-04-27:
We support more radio chips than the SX126x. Plus FHSS is kinda difficult to coordinate in an opportunistinc P2P Mesh Approach.

sachaw, 2022-05-12:
If we could get a comparison between what modules support LR-FHSS contrasting those that we support as a project, it assist us in evaluating the viability.

garthvh, 2022-05-14:
Using FHSS would subject us to a 400ms dwell time limit from the FCC, which would be several times less than our packet size often requires.

closed 2022-05-14 by garthvh