meshtastic/firmware #1423: [Bug]: GPS Longitude has incorrect sign

logikal, 2022-04-29:
### Category

Other

### Hardware

T-Echo

### Firmware Version

1.2.63.9879494

### Description

Longitude has incorrect sign. 

My T-ECHO running 1.2.63.9879494 is displaying and reporting the right numerical digits, but with the wrong sign for the longitude.

Should be *-*122.xx, is 122.xx.

[image]
Incorrect sign on the left, correct sign on the right.

logikal, 2022-04-29:
confirmed that firmware 1.2.62.3ddd74e had the correct longitude sign

caveman99, 2022-05-18:
can you please try https://github.com/meshtastic/artifacts/blob/device/pr/pr1455-firmware-1.2.64.179a262.zip?raw=true

logikal, 2022-05-19:
confirmed fixed in 1.2.64. Thank you!

closed 2022-05-19 by logikal