meshtastic/firmware #1426: Meshtastic Idea

dhubitetackle, 2022-05-01:
Hey

Im wondering if anyone would be able to help me build something. 
Basically i would need a waterproof meshtastic that i would have in one area, and a second unit in another area fitted with an audible alarm. on the first unit, i would press a button, and the unit would save its current gps position. It would then track how far away it moves from that position. if it gets to a set distance away from that locked gps position, that unit alarms, as well as setting off an alarm on the second unit.
It would also need (as a second priority), the ability to chat via apps that are nearby each meshtastic.
The units would need to be battery powered and last at a minimum of overnight, but hopefully several days before needing new charge. They would also need to work at least 200m+ between the 2 meshtastics.
I'd love some help as i am a noob in this area!

Feel free to DM if necessary. Thanks!
Cameron

imkebe, 2022-06-18:
I think more flexible would be implementation of some kind rules/scripts manager. I'm thinking about the same idea however I also want to use external gyroscope module to trigger additional action (message to channel/mqtt)

https://github.com/berry-lang/berry might be an idea. Then we could share recipies for various use cases (like geofencing)

garthvh, 2022-09-28:
This seems like something that would be implemented in a client not on device.

closed 2023-01-04 by sachaw