meshtastic/firmware #1433: [Bug]: BatteryLevel not working on RAK 11200 (ESP32 CPU)

KevinKW87, 2022-05-04:
### Category

Other

### Hardware

Rak11200

### Firmware Version

1.2.63

### Description

Hardware Configuration: RAK11200 + 13300 (1262 Lora) + 12500 (GPS) +19007 (Base Board). 18650 connected to power port. GoalZero Nomad 10 solar panel connected to USB port.

Software configuration: is_always_powered=false. is_router=false. POS_BATTERY=set.

Example position message from unit over MQTT:

{
  "channelId": "LongSlow",
  "gatewayId": "!f71e8b84",
  "packet": {
    "from": -1823672152,
    "to": -1,
    "channel": 177,
    "decoded": {
      "latitudeI": xxx,
      "longitudeI": xxx,
      "altitude": 0,
      **"batteryLevel": 0,**
      "time": 1651685169,
      "locationSource": "LOCSRC_UNSPECIFIED",
      "altitudeSource": "ALTSRC_UNSPECIFIED",
      "posTimestamp": 1651685120,
      "posTimeMillis": 0,
      "altitudeHae": 0,
      "altGeoidSep": 0,
      "pdop": 0,
      "hdop": 0,
      "vdop": 0,
      "gpsAccuracy": 0,
      "groundSpeed": 0,
      "groundTrack": 0,
      "fixQuality": 0,
      "fixType": 0,
      "satsInView": 0,
      "sensorId": 0,
      "posNextUpdate": 0,
      "posSeqNumber": 0
    },
    "id": 2095965523,
    "rxTime": 1651685165,
    "rxSnr": 0,
    "hopLimit": 2,
    "wantAck": false,
    "priority": "BACKGROUND",
    "rxRssi": 0
  }
}

sachaw, 2022-05-12:
@KevinKW87 Could you please confirm if this is a case of the battery not reporting it's % (check the screen),
or is it that the data passed over mqtt is malformed (as it is above).

KevinKW87, 2022-05-14:
Its that the battery isn't reporting its %, I did the malforming to highlight the batteryLevel.

caveman99, 2022-05-16:
In the 11200 Spec there's this definition : #define PIN_VBAT WB_A0 (WB_A0 being 36) - The actual voltage divider of the baseboard is connected to AIN0 so that seems correct.

thebentern, 2022-05-16:
> In the 11200 Spec there's this definition : #define PIN_VBAT WB_A0 (WB_A0 being 36) - The actual voltage divider of the baseboard is connected to AIN0 so that seems correct.

Good catch. I also need to audit the SX1262 pin setup, as I measured very low output on my TinySA compared to the RAK-4631 and even an SX1272 T-Beam.

garthvh, 2022-08-23:
Closing as 1.2 is not getting any updates, please re-test on 1.3 and re-open if it is still an issue.

closed 2022-08-23 by garthvh