meshtastic/firmware #1435: Feature Request: turn BL on/off for router

CHAUPTMANN, 2022-05-05:
Hi, I would like to request the feature, that with a command line command the BL functionality can be turned off and on. This is in particular useful for the nfr52 (RAK4630) hardware, since if used as router, the BL is accessible by anyone without passcode. The is_router true mode turned BL off, but also had negative effects on the LORA connectivity, so is_router doesn't seem to be an option for me.

thebentern, 2022-05-07:
Discussion around this has come up before, my suggestion would be avoid tying it to a router, since some users might want the bluetooth to remain available on a router.
I would propose a preference like `meshastic --set is_bluetooth_disabled true` with a default of false of course.

CHAUPTMANN, 2022-05-08:
This sound like a great proposal.

garthvh, 2022-09-11:
This has been implemented without any connection to mode.

closed 2022-09-11 by garthvh