meshtastic/firmware #1449: [Request]Two pieces of advice about T-echo.

lewisxhe, 2022-05-14:
Hi everyone

https://github.com/meshtastic/Meshtastic-device/blob/12c46ced450c1018638fc038b6c31953077354cc/src/ButtonThread.h#L83

1. It is recommended to use normal `INPUT_PULLUP` for `BUTTON_PIN_TOUCH` instead of `INPUT_PULLUP_SENSE`, because when `BUTTON_PIN_TOUCH` is configured as `INPUT_PULLUP_SENSE`, the capacitive button is used, and this button is easy to be sensed.
In the LilyGo factory firmware, I have changed it to normal input based on the 1.2.60 release.

https://github.com/meshtastic/Meshtastic-device/blob/12c46ced450c1018638fc038b6c31953077354cc/src/gps/GPS.cpp#L41

2. It is recommended to change the GPS communication baud rate to 9600, because if I use other firmware, I will find that the GPS will not be able to communicate, this is due to
Meshtastic uses 4800 as T-echo as the communication rate

The above two points are the problems encountered in the actual production of `T-echo`. I hope they can be changed to help us reduce production errors.

caveman99, 2022-05-14:
1. ButtonThread.h defines two buttons. The line you pointed out is actually driving the touch button at the top of the device, so the INPUT_PULL_SENSE is correct in this case. The user button in the bottom left corner is defined here: https://github.com/meshtastic/Meshtastic-device/blob/12c46ced450c1018638fc038b6c31953077354cc/src/ButtonThread.h#L53-L56 and can indeed be changed to normal behaviour.

2. T-ECHO has a very peculiar GPS setup with a tiny serial buffer that can not be enlarged. You notice that the reduction to 4800 baud is only done for the T-Echo, all other chips are driven at 9600. Also we switch off all unneccessary NMEA Messages to save bandwidth. If we leave the chip at 9600 the UART will lose characters under heavy cooperative multitasking, resulting in flipped signs for coordinates, coordinates and time being decoded wrongly and so on. Despite the missing characters the GPS checksum is somehow decoded right orthe partially decoded 'crap' is not reset on Checksum mismatch and will leak into the next sentence. We were careful not to 'store' these settings to the chip making them permanent, the switch to 4800 baud is (according to the chip spec) not permanent.

lewisxhe, 2022-05-14:
@caveman99 
1. Maybe I'm using the wrong word, I'm talking about the touch button, I don't recommend it as a wake-up function, it's very easy to accidentally touch

2. The GPS module specification does not say that it is a temporary change, nor does it say that it is saved. This is very vague, and I only discovered this problem after passing the actual test. After flashing the `Meshtastic-device` for the first time, It works well, then after I update the firmware to SoftRF, the GPS output baud rate is fixed at 4800, even if I disconnect the battery, it is still fixed at 4800 baud rate after power on again, I have to use 4800 baud rate with GPS communication, and then change to 9600, can read NMEA sentences normally, this is only a suggestion, if the change will reduce the stability, then you can ignore my suggestion.

caveman99, 2022-05-14:
i'll revisit the GPS baud, maybe the reduction in sentences is enough to work reliable. I just know we had trouble with the T-Echo GPS for a long time and that fixed it for us...

As the button is concerned, we have two buttons on the device. the PGM button on the lower left corner (which is wrongly set to SENSE and will be corrected) and the touch button on top next to the GPS antenna (which is really a capacitative touch button and from my understanding needs to be defined as SENSE). Or does the sensor pad on top of the device have a special treatment and not require the SENSE flag?

lewisxhe, 2022-05-17:
If the touch button is set to `SENSE`, then the t-echo is in the off state, and the button can be awakened by touching it, so it is better to set it to the ordinary `INPUT_PULLUP`, which reduces a lot of trouble of accidental touch!

caveman99, 2022-05-17:
I'm just about looking into this, just need to fix the build system first :-)

caveman99, 2022-05-17:
both issues are changed in the code now ...

closed 2022-05-17 by caveman99