meshtastic/firmware #1450: [Bug]:  Set Region by usb not possible

kombjuder, 2022-05-15:
### Category

Other

### Hardware

T-Beam

### Firmware Version

1.2.64

### Description

set region by usb is not posible. The command works without error, but region is not changed.
Tested with t-beam and Heltec
BLE and http works.

thebentern, 2022-05-15:
Which Meshtastic python version was the command issued with?

kombjuder, 2022-05-15:
meshtastic --version
1.2.92

karamo, 2022-05-15:
With CLI 1.2.93 is it possible to set the region parameter:
meshtastic --port com12 --set region EU865
meshtastic --port com12 --get region
region: 3
3 is the index in the list of the regions: {ANZ, CN, EU433, EU865, JP, KR, RU, TW, US, Unset}
The OLED-Screen also shows EU865

kombjuder, 2022-05-16:
EU865 works, EU433 not.

karamo, 2022-05-17:
What does not work?
Look here. All ok.
`Microsoft Windows [Version 10.0.19044.1706]
(c) Microsoft Corporation. Alle Rechte vorbehalten.

C:\Users\KARAHOAN>meshtastic --port com11 --set region EU433
Connected to radio
Set region to EU433
Writing modified preferences to device

C:\Users\KARAHOAN>meshtastic --port com11 --get region
Connected to radio
region: 2
Completed getting preferences

C:\Users\KARAHOAN>`

Also on Boot-Screen **EU433**

thebentern, 2022-05-18:
> What does not work? Look here. All ok.
Same firmware version and python version?

kombjuder, 2022-05-18:
I don’t no what#s going on. I’ve updated all. Then I tried a neu Heltec to EU 865., works. Than I tried an other Heltec to EU433 no error massage but still without region.

So I set regin by ble to EU433 works.

 

I can not say anything more. Sorry, I learned a little programing at a time where a new computer came in a box with a lot of 74XX.

sachaw, 2022-05-18:
We need version info

caveman99, 2022-05-19:
From your nickname i am guessing you're a german speaker? Please run meshtastic --version and meshtastic --info (for every device in question) and paste the results here.

Deinem Nickname entnehme ich, dass du deutschsprachig bist? Bitte führe meshtastic --version und meshtastic --info aus (für jedes in Frage kommende Gerät) und füge die Ergebnisse hier ein.

kombjuder, 2022-05-19:
Hallo Thomas,

 

ein kompletter Mitschnitt.

Bei einem davor geschriebenen T->Beam ist das nicht aufgetreten. Da konnte die „Region“ gesetzt werden.

 

 

Ich hoffe, die Infos reichen.

 

Requirement already satisfied: pycparser in d:\users\k-h\meshtastic-flasher\venv\lib\site-packages (from cffi>=1.12->cryptography>=2.1.4->esptool->meshtastic-flasher) (2.21)

pypi_version:1.0.105

running: 1.0.105

start of detect

Detected Heltec

Detected Heltec

Detected Heltec

Detected Meshtastic DIY

Detected T-Lora

User said no to device having Meshtastic on the device so no detection using the Meshtastic python method

filenames:['1.2.64.fc48fcd\\firmware-heltec-v1-1.2.64.fc48fcd.bin', '1.2.64.fc48fcd\\firmware-heltec-v2.0-1.2.64.fc48fcd.bin', '1.2.64.fc48fcd\\firmware-heltec-v2.1-1.2.64.fc48fcd.bin', '1.2.64.fc48fcd\\firmware-meshtastic-diy-v1-1.2.64.fc48fcd.bin', '1.2.64.fc48fcd\\firmware-nano-g1-1.2.64.fc48fcd.bin', '1.2.64.fc48fcd\\firmware-rak11200-1.2.64.fc48fcd.bin', '1.2.64.fc48fcd\\firmware-tbeam-1.2.64.fc48fcd.bin', '1.2.64.fc48fcd\\firmware-tbeam0.7-1.2.64.fc48fcd.bin', '1.2.64.fc48fcd\\firmware-tlora-v1-1.2.64.fc48fcd.bin', '1.2.64.fc48fcd\\firmware-tlora-v2-1-1.6-1.2.64.fc48fcd.bin', '1.2.64.fc48fcd\\firmware-tlora-v2-1.2.64.fc48fcd.bin', '1.2.64.fc48fcd\\firmware-tlora_v1_3-1.2.64.fc48fcd.bin']

self.detected_list:['heltec-v2.1', 'heltec-v1', 'heltec-v2.0', 'meshtastic-diy-v1', 'tlora-v1-3']

User picked radio heltec-v2.0

self.device_from_picker:heltec-v2.0

index:3

end of detect

Flash was clicked.

update only is not checked

User confirmed they want to flash

do esptool update_only:False

finished:True

 

 

PS C:\Users\k-h> meshtastic --info

Connected to radio

 

Owner: Unknown 2d84 (?84)

My info: { "myNodeNum": 302329220, "numBands": 13, "firmwareVersion": "1.2.64.fc48fcd", "rebootCount": 2, "bitrate": 17.08847, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true }

Nodes in mesh:  {'num': 302329220, 'user': {'id': '!12052d84', 'longName': 'Unknown 2d84', 'shortName': '?84', 'macaddr': 'a4:cf:12:05:2d:84', 'hwModel': 'HELTEC_V2_0'}, 'position': {}}

 

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300 }

 

Channels:

  PRIMARY psk=default { "modemConfig": "Bw125Cr48Sf4096", "psk": "AQ==" }

 

Primary channel URL: https://www.meshtastic.org/d/#CgUYAyIBAQ

 

PS C:\Users\k-h> meshtastic --port com6 --set region EU433

Connected to radio

Set region to EU433

Writing modified preferences to device

 

PS C:\Users\k-h> meshtastic --port com6 --get region

Connected to radio

region: 0

Completed getting preferences

karamo, 2022-05-19:
> > What does not work? Look here. All ok.
> 
> Same firmware version and python version?

As I wrote befor: MT-Python 1.2.93 / FW 1.2.64

caveman99, 2022-05-20:
@kombjuder Der Heltec ist etwas zickig, wenn man ihn von Windows aus programmiert. Beobachte bitte mal, ob der neu startet (einen Reset ausführt) wenn du den --set Befehl absetzt. In dem Fall könnte es sein, daß er nicht genug Zeit hat, die Einstellung in den Speicher zu schreiben bevor er neu startet. Ein paar Wiederholungen des Befehls führen dann zum Erfolg. Ändern können wir da leider nichts, das liegt daran, wie der serielle USB-Chip auf dem Board verschaltet ist.

The Heltec is a bit touchy when set from Windows. Please observe if it restarts (resets) when you issue the --set command. If so, it may not have enough time to write the setting to memory before restarting. A few repetitions of the command will then lead to success. Unfortunately, there is nothing we can do about this, it is due to the way the serial USB chip is wired on the board.

garthvh, 2022-07-03:
1.2 Issue

closed 2022-07-03 by garthvh