meshtastic/firmware #1451: [Feature Request]: nRF52 support for Range Test module

charminULTRA, 2022-05-15:
I would like to see nRF52 support for the Range Test module, especially as these devices are becoming more popular. Additionally, since the stock PCB antenna for the RAK 4631 is a PCB antenna with only uFL connector, typical VSWR measurement tools cannot be used without special adapters.

mc-hamster, 2022-05-24:
> I would like to see nRF52 support for the Range Test module, especially as these devices are becoming more popular. Additionally, since the stock PCB antenna for the RAK 4631 is a PCB antenna with only uFL connector, typical VSWR measurement tools cannot be used without special adapters.

What would be the minimum support you'd like to see by the Range Test Module on the RAK?

charminULTRA, 2022-05-24:
> What would be the minimum support you'd like to see by the Range Test Module on the RAK?

I think Sender functionality would be a good minimum, since testing send capability/range would probably be more of a question mark than receive.  Additionally, I imagine the File Browser access for rangetest.csv (for the Receive role) might be more complicated/time-consuming to implement without WiFi, but I'm just speculating as a non-technical person.

mc-hamster, 2022-05-24:
Sender can be added as a minimum. Sure. That should be possible.

alistair23, 2023-05-07:
The entire range test module is ESP32 only: https://github.com/meshtastic/firmware/blob/master/src/modules/esp32/RangeTestModule.cpp

But the code also contains `#ifdef ARCH_ESP32`. Could the file not be built for all targets (with some changes to the current code)? That should be a good start towards send support on other platforms.

PR #2534 (2023-06-01): Enable range test module (sending only) on NRF
Closes #1451
closed 2023-06-01 by thebentern