meshtastic/firmware #1460: [Bug]: LILYGO® TTGO Lora v1.0 does not enter low power mode (says USB when unplugged)

scottsweb, 2022-05-21:
### Category

Hardware Compatibility

### Hardware

T-Lora v1

### Firmware Version

1.2.6.4.fc48fcd

### Description

This board is not correctly detecting when it is unplugged from USB. I have a LiPO plugged in and when I remove the USB and stop charging, Meshtastic says `USB` on the screen. The device also never enters its low power state, with the screen staying on indefinately.

caveman99, 2022-05-21:
have you set is_always_powered?

scottsweb, 2022-05-21:
It is currently using all default settings. `is_always_powered` is off.

scottsweb, 2022-05-21:
Just testing `1.2.6.0` and it is behaving the same too.

caveman99, 2022-05-21:
mh, not sure the TTGO 1.0 has a ADC pin for battery sampling...

caveman99, 2022-05-21:
https://github.com/meshtastic/Meshtastic-device/blob/master/variants/tlora_v1/variant.h - the 1.0 does nt have a BATTERY_PIN so there's no way of telling if it is powered or not. This was introduced in v1.3 of the board.

garthvh, 2022-05-21:
Yeah this is a know issue with the hardware

scottsweb, 2022-05-21:
Ok, that makes sense! thanks for the fast replies. Is there a setting I can set to force it into a low power mode after a short while?

caveman99, 2022-05-21:
i think in 1.2 is_router will do that.

garthvh, 2022-05-21:
Low power turns off bluetooth and wifi quickly, so if you are using the device with a client application it is problematic.

closed 2022-05-21 by garthvh

scottsweb, 2022-05-21:
Ahh interesting, I had no idea they were the same setting. `Powered by Low Power (solar)` seems to be turning the screen off, but as @garthvh, this might not be ideal for a client. I will keep tinkering. 

Thanks again for all your replies.

garthvh, 2022-05-21:
The is always power setting was created because the detection on these boards is bad without a battery pin, they tend to work best powered by USB.

garthvh, 2022-05-21:
duplicate of #581