meshtastic/firmware #1470: [Bug]: Solar powered T-Beam V1.1 not waking up after battery run down from no sun

chopstar001, 2022-05-24:
### Category

Other

### Hardware

T-Beam

### Firmware Version

1.2.59.d81c1c0

### Description

Solar powered T-Beam v1.1 (set to isRouter true) not waking up after having all its power drained to completely flat battery from lack of sunlight.
Later having to manually power device back on, and often battery is found to have fully recharged.

Solar Panel:
5v 4w 800ma 
Power Management:
5V 2A Solar Panel Power Bank USB Charge Voltage Controller Regulator.

closed 2022-05-24 by mc-hamster

caveman99, 2022-05-24:
the AXP does not have a setting to switch power back on when it returns? Major design flaw, every 80s bios can do that ;-)

mc-hamster, 2022-05-24:
> the AXP does not have a setting to switch power back on when it returns? Major design flaw, every 80s bios can do that ;-)

I’ve looked and so have many others.

A watchdog can be added to the power on input, but that’s a board level design decision.

geeksville, 2022-05-25:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/solar-node-mounted-in-a-tree/5283/35

chopstar001, 2022-06-01:
Just wanted to add, that for me, a work around was to change the super deep sleep setting, to wake from this state every 2 hours (might be better set at 6+ hours to give it time to accumulate charge) instead of the default 1 year. (meshtastic --set sds_secs 7200). Otherwise its like Snow White waiting for a Prince to come along to push her buttons. Tested this setting for the last few nights making sure it ran flat by nightfall, and found it was waking up in the morning.