meshtastic/firmware #1482: 1.3 is_always_powered Settings Cleanup

garthvh, 2022-06-02:
### Category

Other

### Hardware

T-Lora v1, T-Lora v1.3, Heltec v1, Heltec v2, Heltec v2.1

### Firmware Version

1.3+

### Description

Me and @thebentern worked to add is_always_powered when we first starting working with meshtastic, and then @mc-hamster implemented is_low_power as a way fix the need to manually push the user button on tbeams to wake them back up.

is_always_powered is now being used for a number of unintended purposes and creating user confusion.

This setting is being used by two groups currently, people with devices without battery pins (heltec, lora32 v1) and people who want the screen to never turn off.

I would propose we add a setting to keep the screen on, and delete the is_always_powered setting and prevent devices without a battery pin from sleeping as we can't do any battery detection.

garthvh, 2022-06-12:
is_low_power and is_power_saving seem to do the same thing, and both are questionable as user settings.

mesh_Sds_Timeout_Secs, sds_Secs and ls_Secs seem like things that should be sensible defaults instead of settings

garthvh, 2022-06-13:
https://github.com/meshtastic/Meshtastic-device/pull/1507

garthvh, 2022-06-16:
These are all the power settings

charge_current
is_power_saving
is_always_powered
on_battery_shutdown_after_secs
adc_multiplier_override
wait_bluetooth_secs
mesh_sds_timeout_secs
sds_secs
ls_secs
min_wake_secs

mc-hamster, 2022-06-17:
Good list @garthvh 

add_multiplier_override is used for the battery monitor, that needs to stay.

is_power_saving and is_always_powered is a hack and can be rolled into device roles or at the very least, renamed to describe exactly what it does.

how can values for these be defined for roles?

caveman99, 2022-06-17:
from the device code the ls_secs is (was?) needed by the apps to know when the bluetooth might go offline. Default value is always sent on the PhoneAPI regardless of being actually set. the other timing settings i'd have to dig through as well.

the adc_multiplier is here to stay. And we need a way to either figure out automatically or have a settings override when a device either has no battery sensing pin at all or when it has but there is no battery connected. 

Idea to assess the latter would be by looking at the changes over the measurings and if they were too rapid (e.g. more than 5% in 20 seconds) the pin was assumed floating. And set the info to 'no battery present'

andrekir, 2022-06-17:
maybe worth mentioning devices don't light sleep anymore enabling `is_power_saving` (since 1.2)
`mesh_sds_timeout_secs`: going into super deep sleep for a year after 2 hours of mesh inactivity by default is a trouble maker.

garthvh, 2022-06-17:
>  how can values for these be defined for roles?

Feels like they are mostly in the role already as least as far as the actual intent, they do get used a lot when users are flailing a bit, but I think that mostly make things worse.

* Client 
* ClientMute: Packets will not hop over this node, does not contribute to routing packets for mesh.
* Router - Mesh packets will prefer to be routed over this node. This node will not be used by client apps. The wifi/ble radios and the oled screen will be put to sleep.
* Router Client - Mesh packets will prefer to be routed over this node. The Router Client can be used as both a Router and an app connected Client."

garthvh, 2022-06-17:
> from the device code the ls_secs is (was?) needed by the apps to know when the bluetooth might go offline. Default value is always sent on the PhoneAPI regardless of being actually set. 

We deleted this functionality and made is_power_saving if anyone wanted to go back, which nobody did, so I think both can go away completely.

caveman99, 2022-06-17:
Good more crud that can be removed. I just coded this for the new segmented config ;-)

andrekir, 2022-06-19:
> part of the removed "device is connected but sleeping" tbeam functionality in the android app

small clarification, this was just fixed to show correctly when `ls_power_saving` is enabled. functionality is still there and "works as designed", though devices don't go into sleep (could be fixed?). I'd prefer to leave it in, but not opposed to any other direction.

devices would "sleep" for `ls_secs` then "stay awake" for `wait_bluetooth_secs`, so the pair can be considered.

garthvh, 2022-06-19:
@andrekir Is the is_power_setting the one you have left in the android app at this point?

garthvh, 2022-06-20:
Closing this for now, we can revisit if we find later any of the remaining settings conflict with roles as they are implemented

closed 2022-06-20 by garthvh