meshtastic/firmware #1512: [Board]: LoPy4

cabinlab, 2022-06-14:
### SOC

ESP32

### Lora IC

SX1276

### Product Link

https://pycom.io/product/lopy4/

### Description

No GPS
Tested on 1.2.65. Not yet tested on 1.3_
[Forum Post](https://meshtastic.discourse.group/t/pycom-lopy4-basic-functionality/5993/1)

mc-hamster, 2022-06-22:
How many people are using this board?

garthvh, 2022-08-04:
End of life and no other interest, can revisit if next version gets some traction

closed 2022-08-04 by garthvh

gretel, 2024-06-01:
works for me, taken from the [discourse thread](https://meshtastic.discourse.group/t/pycom-lopy4-basic-functionality/5993):

`variant.h`

```c
#undef GPS_RX_PIN
#undef GPS_TX_PIN

#define I2C_SDA 12
#define I2C_SCL 13

#undef PIN_NEOPIXEL // no support
#define LED_PIN 16

#define BUTTON_NEED_PULLUP
#define BUTTON_PIN 17 // pycom expansion board v2

// Pin definition in LoPy4 Manual is incorrect. See link to errata:
// https://forum.pycom.io/topic/3403/wiring-of-dio-pins-of-lora-sx127x-chip-to-esp32/4?_=1655162492722
// https://forum.pycom.io/topic/3483/erratum-in-lopy4-specification-document
// LoPy (original, not 4) pin definition:
// https://forum.pycom.io/post/20460

#define USE_RF95
#define LORA_DIO0 23  // Note: All DIO are wired through a diode bridge to a single pin
#define LORA_RESET -1 // NC
#define LORA_DIO1 23  // Not really used
#define LORA_DIO2 23  // Not really used

#define LORA_RXEN -1
#define LORA_TXEN -1

#define RF95_RXEN LORA_RXEN
#define RF95_TXEN LORA_TXEN

#undef RF95_SCK
#define RF95_SCK 5
#undef RF95_MISO
#define RF95_MISO 19
#undef RF95_MOSI
#define RF95_MOSI 27
#undef RF95_NSS
#define RF95_NSS 18
```

`platformio.ini`

```ini
; Meshtastic Pycom Lopy4
[env:pycom-lopy4]
board = lopy4
extends = esp32_base
build_flags =
  ${esp32_base.build_flags}
  -D PRIVATE_HW 
  -I variants/pycom-lopy4
```