meshtastic/firmware #1549: 1.3 Feature Request: Channel QR code URL Get / Set methods on device

garthvh, 2022-07-03:
As a developer I just want to get a channel URL from the device, or set the channel settings from a channel URL I have scanned, I don't want to have to additionally parse all the settings out on each platform or base64 encode anything.

* [ ] Create an admin message to SET channel settings from a URL (or just the relevant after hash portion)
* [ ] Create a way to GET the channel url from the device (or just the relevant after hash portion)

raldi, 2022-07-18:
I can take a crack at this. Do you want to spec out the API endpoints for these two messages or should I just pick something that seems to match the surrounding code?

garthvh, 2022-07-18:
You know C better than me @mc-hamster / @caveman99 any endpoint style requests?

mc-hamster, 2022-07-18:
Hmm... this may limit the size of that payload. 

There's up to 37% overhead with base64 and that would then be packed into one of our payloads. 

I think this will be ok, but may lead to a bug down the line if we are not mindful of the configuration packet size. 

No preference on how it's implemented. Creative license and all that :)

garthvh, 2022-07-18:
The 1.3 url uses a capital E in the android app, so that is what I have set up on iOS with the apple applink (which passes the url from the camera to the app)

garthvh, 2022-07-27:
The admin message has a number of get / set commands for owner, channels, config eat,I would imagine like that.

sachaw, 2022-07-28:
I'm all for simplifying this, however I believe making the device generate the url would be slow or limiting, take the qr component of the webUI for example: https://github.com/meshtastic/meshtastic-web/blob/a2280ab96e73a6fda2dfc6652e9e74e037299605/src/components/Dialog/QRDialog.tsx#L36-L50
It allows me to dynamically generate QR codes/URL's based on what the user wants to include, in very little code.

garthvh, 2022-07-31:
@sachaw am I missing where there is any implemented functionality for channels other than primary and admin?

No packets are sent using any other channels are they?

sachaw, 2022-07-31:
I have support to send any packet on any channel, the webui supports setting crypto and messaging on all channels too.
https://client.meshtastic.org/

garthvh, 2022-07-31:
> I have support to send any packet on any channel, the webui supports setting crypto and messaging on all channels too.
> https://client.meshtastic.org/

This winds up only being supported only in the web ui then right? No messaging on any channel other than primary  flows through the text message app used by every other client.

sachaw, 2022-07-31:
It should be trivial to support, when sending messages, you just specify what channel index in the meshpacket, and received packets have that property set, so you can classify them.

I'd say it's an important feature as people can have multiple private channels with different crypto

garthvh, 2022-07-31:
I don't think it is trivial or it would have been implemented in the text message app when the multiple channels were added to the firmware two years ago.  Also seems like a huge UX change to subject users too while we are trying to get 1.3 out. Does this mean users can DM on specific channels and the UX will need to support duplicate "contacts" because they are on separate channels?

sachaw, 2022-07-31:
Maybe we're thinking of these features differently.

The way I think the separation should be done is as follows:
When the device is configured, take the max channels reported and request each of them.
For each enabled channel, create a new chat instance in your state machine.
When messages are sent or received, take the packets channel number and push it to the appropriate message store. All of this functionality is already implemented.

As for DM's there was quite a bit of discussion on it before and we decided not to bother with it for now, there is already firmware level support for DM's but just to keep features lean we opted not to implement it in the apps.

sachaw, 2022-07-31:
As for "contacts" the nodes is built from packets broadcast on the default channel, so it shouldn't have any bearing here.

garthvh, 2022-08-03:
> Maybe we're thinking of these features differently.
>
> The way I think the separation should be done is as follows: When the device is configured, take the max channels reported and request each of [...]

Closing this issue, made a new issue with your comment I will add too.

closed 2022-08-03 by garthvh