meshtastic/firmware #1552: Feature Request: Ability to message outside primary channel

VladTheSkewerer, 2022-07-05:
Didn't see this explicitly listed, but it's probably on a roadmap somewhere.

Can we get a method for selecting a channel other than the primary for messaging?  It would also be great to have a way to make it sticky so that channel is selected when the app is restarted, though that's just secondary.

Thanks for all your work!

garthvh, 2022-07-05:
[Discourse discussion](https://meshtastic.discourse.group/t/encryption-secondary-channels/4838/19)

garthvh, 2022-07-05:
Related pull request that is still consuming too much memory https://github.com/meshtastic/Meshtastic-device/pull/1509

garthvh, 2022-07-05:
> I would also like to see this. The ability to have a Public, Private and Admin Channels accessible from the app would be killer. It would be really nice to perhaps have a --ch-add functionality with --ch-index and PSK options at time of channel creation(in the app). All in good time, thanks for everyone's hard work on this project!

The admin channel sends config commands, why would users want this to be a messaging capable channel in a phone app?

phazed0, 2022-07-06:
> The admin channel sends config commands, why would users want this to be a messaging capable channel in a phone app?

Sorry, I'm new here - didn't know there was a Remote Node admin channel literally named admin. In my previous example make it Public Chat, Private Chat, Private Chat 2. Simply the ability to create and pick between 'known' channels in the app.

garthvh, 2022-07-06:
> Sorry, I'm new here - didn't know there was a Remote Node admin channel literally named admin. In my previous example make it Public Chat, Private Chat, Private Chat 2. Simply the ability to create [...]

Those sound like private groups and not channels.

garthvh, 2022-09-11:
The device part of this is all in place, this is completely dependent on clients.  DM's are only be available on the primary channel and there can generally be up to 6 private chat groups if your preset offers a max of 8 channels

closed 2022-09-11 by garthvh