meshtastic/firmware #1558: [Bug]: 1.3.25 BME280 sensor temperature is too high ( Sensor Thermal Control )

0xcodedead, 2022-07-15:
### Category

Other

### Hardware

T-Echo

### Firmware Version

1.3.25

### Description

I'm using a T-ECHO and BME280 sensor and the firmware is 1.3.25a.

There is a problem that the BME280 sensor comes out 5-7 degrees higher than the actual temperature. There are significant differences from the values ​​​​of two or more other thermometers.
 
After reviewing this issue, it appears that the initial setup of the BME280 is causing the sensor to trigger too frequently, causing it to self-heat.

So, it seems like you need to add code to control the heat by changing the initial settings in the BME280's firmware. The code is in the link below.

Also a discussion of issues and reference codes for the BME280 can be found at the link below.

1. https://www.esp8266.com/viewtopic.php?f=13&t=8030&sid=5b736d896eb9758f6ae56ce5aefbaceb&start=28
2. https://www.kandrsmith.org/RJS/Misc/Hygrometers/calib_many.html
3. https://github.com/opendata-stuttgart/sensors-software/blob/619ef6fab6a611d838bd7a71a29ce16f204b7eef/airrohr-firmware/airrohr-firmware.ino#L3551

0xcodedead, 2022-07-15:
According to the first article, you can apply the code below.

// Adafruit_BME280 bme

1. After initializing the library

bme.setSampling(Adafruit_BME280::MODE_FORCED, // takeForcedMeasurement must be called before each reading
Adafruit_BME280::SAMPLING_X1, // Temp. oversampling
Adafruit_BME280::SAMPLING_X1, // Pressure oversampling
Adafruit_BME280::SAMPLING_X1, // Humidity oversampling
Adafruit_BME280::FILTER_OFF,
Adafruit_BME280::STANDBY_MS_1000);

2. Before each reading:
bme.takeForcedMeasurement();

caveman99, 2022-11-05:
This is part of develop branch now...

https://github.com/meshtastic/firmware/commit/3d9633a56cbf4520c81926d9229053563123657f

wonder if the 680 chips need the same tratment.

caveman99, 2022-11-06:
The 680 is fine and uses forced measurement already. However i changed the init a bit to use less power. And take only one reading while sampling and not 4.

closed 2022-11-06 by caveman99

FAte-CzE, 2024-06-13:
Hi, I'm experiencing this issue on 2.3.11 now