meshtastic/firmware #1562: [Bug]: t-beam 0.7 reboot loop

dikkedimi, 2022-07-20:
### Category

Other

### Hardware

T-Beam 0.7

### Firmware Version

1.3.25.4de6d5bd-d

### Description

Hi All!

Trying to get things running on the bare minimum I could get, which is a t-beam 0.7 for now.

With the stock git repo built in Visual studio code (I really do prefer atom, but alas) Building and uploading worked after a few issues detecting the serial port (have to reboot everytime I unplug the t-beam for it to be detected). Anyhow, I got it to upload the compiled code. But then it just went into reboot loops. (see relevant log)

I tried to get some info by decoding the stack trace output.
```
Output:
0x401f0c4c: OLEDDisplay::drawVerticalLine(short, short, short) at /Volumes/HDD/Git/Meshtastic-device/.pio/libdeps/tbeam0.7/ESP8266 and ESP32 OLED driver for SSD1306 displays/src/OLEDDisplay.cpp:455
0x400f427b: OLEDDisplay::drawRect(short, short, short, short) at /Volumes/HDD/Git/Meshtastic-device/.pio/libdeps/tbeam0.7/ESP8266 and ESP32 OLED driver for SSD1306 displays/src/OLEDDisplay.cpp:210
0x400dcaab: graphics::Screen::adjustBrightness() at /Volumes/HDD/Git/Meshtastic-device/src/graphics/Screen.cpp:1621
0x400ddb89: concurrency::ButtonThread::userButtonPressedLong() at /Volumes/HDD/Git/Meshtastic-device/src/ButtonThread.h:131
0x400f566e: OneButton::tick(bool) at /Volumes/HDD/Git/Meshtastic-device/.pio/libdeps/tbeam0.7/OneButton/src/OneButton.cpp:305
0x400f56de: OneButton::tick() at /Volumes/HDD/Git/Meshtastic-device/.pio/libdeps/tbeam0.7/OneButton/src/OneButton.cpp:199
0x400ddbc5: concurrency::ButtonThread::runOnce() at /Volumes/HDD/Git/Meshtastic-device/src/ButtonThread.h:93
0x400d53fe: concurrency::OSThread::run() at /Volumes/HDD/Git/Meshtastic-device/src/concurrency/OSThread.cpp:78
0x400f6519: ThreadController::runOrDelay() at /Volumes/HDD/Git/Meshtastic-device/.pio/libdeps/tbeam0.7/Thread/ThreadController.cpp:59
0x400dded4: loop() at /Volumes/HDD/Git/Meshtastic-device/src/main.cpp:479
0x40129345: loopTask(void*) at /Users/user/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19
```
Unfortunately I don't really know what any of it means or where to begin mitigating this issue

tried again flashing according to the instructions (why are there only instructions for windows???)
```
Mixing-Desk:~ User$ /Users/user/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool --chip esp32 --port /dev/cu.wchusbserial54350181341 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /Users/user/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/partitions/boot_app0.bin 0x1000 /Users/user/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/bin/bootloader_qio_80m.bin 0x10000 /Volumes/HDD/Git/LilyGo-LoRa-Series/firmware/Meshtastic/firmware-tbeam0.7-EU865-1.0.0.bin 
esptool.py v3.0-dev
Serial port /dev/cu.wchusbserial54350181341
Connecting.....
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 78:21:84:87:e1:88
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 17683.9 kbit/s)...
Hash of data verified.
Compressed 18656 bytes to 12053...
Wrote 18656 bytes (12053 compressed) at 0x00001000 in 0.1 seconds (effective 1054.7 kbit/s)...
Hash of data verified.
Compressed 1133824 bytes to 664220...
Wrote 1133824 bytes (664220 compressed) at 0x00010000 in 9.9 seconds (effective 913.1 kbit/s)...
Hash of data verified.
```
But I got a boot loop (again) with a lot of unkown characters in serial.

### Relevant log output

```Shell
ets Jul 29 2019 12:21:46
[... 129 lines cut ...]
PC      : 0x401f0c4c  PS      : 0x00060630  A0      : 0x800f427e  A1      : 0x3ffd5e10  
A2      : 0x3ffd6378  A3      : 0x00000180  A4      : 0x00000002  A5      : 0x00000004  
A6      : 0x3ffc2f68  A7      : 0x3ffc2ef4  A8      : 0x000000c0  A9      : 0x00000001  
A10     : 0xffffffff  A11     : 0x3ffc2f70  A12     : 0x400ddbb4  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3f400544  SAR     : 0x00000002  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000180  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x401f0c4c:0x3ffd5e10 0x400f427b:0x3ffd5e30 0x400dcaab:0x3ffd5e50 0x400ddb89:0x3ffd5e80 0x400f566e:0x3ffd5ea0 0x400f56de:0x3ffd5ec0 0x400ddbc5:0x3ffd5ee0 0x400d53fe:0x3ffd5f00 0x400f6519:0x3ffd5f20 0x400dded4:0x3ffd5f50 0x40129345:0x3ffd5f70

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:728
load:0x40078000,len:9132
load:0x40080400,len:5100
entry 0x40080640
```
```

dikkedimi, 2022-07-20:
After soldering on the oled display the bootloop stops.

I am now able to connect to the Wifi AP. But unable to change any settings or add the radio to my meshtastic app on android 11. The app just spins the indicator after pressing the + button.

caveman99, 2022-07-20:
??:??:?? 0 I2C device found at address 0x34

That would be the first T-BEam 0.7 with an AXP192 power chip. Try the "tbeam" firmware with is for 1.0 and newer

closed 2022-07-20 by caveman99

dikkedimi, 2022-07-20:
> Try the "tbeam" firmware which is for 1.0 and newer

@caveman99 Thanks for your reply!
I have flashed the t-beam 1.0 variant to the device. Same info on the display. Able to connect to the AP but that's it. not able to configure anything. Alignment of some display items is on top of the progress bar. Doesn't really seem to do anything different.

caveman99, 2022-07-20:
1.3 is still very much Work in Progress. You got the right firmware type now, for a good user experience yu might want to reflash to 1.2.x

feh123, 2022-07-20:
I have seen something like this on my tbeams. I had 1.3.25 running well on two tbeams T22 V1 20191212 SX1262 and same same version with a LORA32 radio. When I upgraded to 1.3.26 the SX1262 was fine but the LORA32 kept just cycling the opening Meshtastic Logo page.

dikkedimi, 2022-07-20:
@caveman99 You seem to be right, after compiling and uploading the latest 1.2 release (zipped sourcecode), and selecting t-beam instead of t-beam 0.7 (this should maybe be mentioned on the documentation page to avoid frustrating users?), I now have it working :D

Althoug I am not able to create a new channel but at least I got connection with my phone

dikkedimi, 2022-07-20:
Hmm, no problems with my android phone, but iOS compains about 1.2 firmware not being supported. Too bad.

caveman99, 2022-07-20:
1.2 only works with 1.2 clients and 1.3 only with 1.3 - with iOS you need to select the other release (1.2) in testflight. 1.2 is not a great experience with iOS though, because of the older BLE stack. That's the reason we ripped that out and put a new one in 1.3

dikkedimi, 2022-07-20:
As I now have both OS'es to my avail, I'd love to support any testing. For now I am very happy with the support getting things going. I'll have to check in once my 433mhz units arrive on the slow boat.

Btw, is the point of using wi-fi instead of bluetooth in the 1.3 clients also paving the way for multi-phone pairing? I'd love to have both my phones connect to the node when in range.

garthvh, 2022-07-20:
The main use for wifi is to be a base station or bridge the network to the internet, wifi connected devices are much more power hungry and not a great choice as a portable handset an esp32 with WiFi takes 200mah while a nrf52 device uses 5-15ma.  Multi user is an open enhancement, but would be a lot of work and is likely a device memory problem.