meshtastic/firmware #1564: [Bug]: 1.3.26 - t-echo - throws critical fault #7 after connecting phone and renaming, resets, then boots with default name

Vertikar, 2022-07-22:
### Category

BLE, Other

### Hardware

T-Echo

### Firmware Version

1.3.26.0010231

### Description

Have had device clear its settings twice after renaming it, then seems to throw critical error 7.
Most recent attempt of setting name didn't seem to trigger the reset and setting clear, but still threw critical error 7

### Relevant log output

```Shell
08:09:15 554 [Router] FIXME-update-db Sniffing packet
08:09:15 554 [Router] Delivering rx packet (id=0x1fc32173 Fr0x0f To0x0f, WantAck0, HopLim3 Ch0x0 Portnum=6 requestId=ef2d4058 rxtime=1488269354 priority=70)
08:09:15 554 [Router] Forwarding to phone (id=0x1fc32173 Fr0x0f To0x0f, WantAck0, HopLim3 Ch0x0 Portnum=6 requestId=ef2d4058 rxtime=1488269354 priority=70)
08:09:15 554 [Router] Update DB node 0xc65fc90f, rx_time=1488269354
08:09:15 554 [Router] Module 'routing' considered
08:09:15 554 Telling client we have new packets 10
08:09:15 554 BLE notify fromNum
08:09:15 554 [Screen] getFromRadio, state=7
08:09:15 554 [Screen] phone downloaded packet (id=0x1fc32173 Fr0x0f To0x0f, WantAck0, HopLim3 Ch0x0 Portnum=6 requestId=ef2d4058 rxtime=1488269354 priority=70)
08:09:16 555 [PowerFSM] toRadioWriteCb data 0x2001bd98, len 28
08:09:16 555 [PowerFSM] PACKET FROM PHONE (id=0xef2d4059 Fr0x00 To0x0f, WantAck1, HopLim0 Ch0x0 Portnum=6 priority=70)
08:09:16 555 [PowerFSM] Enqueued local (id=0xef2d4059 Fr0x00 To0x0f, WantAck1, HopLim0 Ch0x0 Portnum=6 rxtime=1488269356 priority=70)
08:09:16 555 [Router] Add packet record (id=0xef2d4059 Fr0x00 To0x0f, WantAck1, HopLim0 Ch0x0 Portnum=6 rxtime=1488269356 priority=70)
08:09:16 555 [Router] handleReceived(REMOTE) (id=0xef2d4059 Fr0x00 To0x0f, WantAck1, HopLim0 Ch0x0 Portnum=6 rxtime=1488269356 priority=70)
08:09:16 555 [Router] Module 'Admin' wantsPacket=1
08:09:16 555 [Router] Received Admin from=0x0, id=0xef2d4059, portnum=6, payloadlen=6
08:09:16 555 [Router] Client is setting the config
08:09:16 555 [Router] Setting config: LoRa
08:09:16 555 [Router] Expanding short PSK #1
08:09:16 555 [Router] Wanted region 6, using ANZ
08:09:16 555 [Router] Set radio: region=ANZ, name=LongF, config=0, ch=39, power=30
08:09:16 555 [Router] Radio myRegion->freqStart / myRegion->freqEnd: 915.000000 -> 928.000000 (13.000000 mhz)
08:09:17 555 [Router] Radio myRegion->numChannels: 52
08:09:17 555 [Router] Radio channel_num: 39
08:09:17 556 [Router] Radio frequency: 10790.000000
08:09:17 556 [Router] Slot time: 42 msec
08:09:17 556 [Router] NOTE! Recording critical error 7 at src/mesh/SX126xInterface.cpp:91
08:09:17 556 [Router] Can't write prefs
08:09:17 556 [Router] Can't write prefs
08:09:17 556 [Router] Can't write prefs
08:09:17 556 [Router] Can't write prefs
08:09:17 556 [Router] Module 'Admin' considered
08:09:17 556 [Router] Module 'routing' wantsPacket=1
08:09:17 556 [Router] Received routing from=0x0, id=0xef2d4059, portnum=6, payloadlen=6
08:09:17 556 [Router] Routing sniffing (id=0xef2d4059 Fr0x00 To0x0f, WantAck1, HopLim0 Ch0x0 Portnum=6 rxtime=1488269356 priority=70)
08:09:17 556 [Router] Alloc an err=0,to=0xc65fc90f,idFrom=0xef2d4059,id=0x1fc32174
08:09:17 556 [Router] Enqueued local (id=0x1fc32174 Fr0x0f To0x0f, WantAck0, HopLim0 Ch0x0 Portnum=5 requestId=ef2d4059 rxtime=1488269357 priority=120)
08:09:18 557 [Router] FIXME-update-db Sniffing packet
08:09:18 557 [Router] Module 'routing' considered
08:09:18 557 [Router] Add packet record (id=0xef2d405a Fr0x00 To0x0f, WantAck1, HopLim0 Ch0x0 Portnum=6 WANTRESP rxtime=1488269356 priority=70)
08:09:18 557 [Router] handleReceived(REMOTE) (id=0xef2d405a Fr0x00 To0x0f, WantAck1, HopLim0 Ch0x0 Portnum=6 WANTRESP rxtime=1488269358 priority=70)
```

garthvh, 2022-07-22:
Try the newest build 26 contains a radio bug

Vertikar, 2022-07-22:
Just saw that in discord, I'll give that a go and report back

Vertikar, 2022-07-22:
That seems to have fixed it, thanks

closed 2022-07-22 by Vertikar