meshtastic/firmware #1565: [Bug]: After trying to enable BME280 Sensor via iOS App, T-Echo is in reboot loop

arnotron, 2022-07-24:
### Category

Other

### Hardware

T-Echo

### Firmware Version

1.3.26.0010231

### Description

When I realised that my T-Echo had not yet enabled the BME280 Sensor and I tried to enable it via the iOS app, it got stuck in a reboot loop. 

Serial Console reports an error writing prefs. Is there a way to set the configuration back to default? Downgrading to the previous version 1.3.25.85f46d3 did not change anything.

### Relevant log output

```Shell
��H??:??:?? 1

//\ E S H T /\ S T / C

??:??:?? 1 Filesystem files:
??:??:?? 1   radio.proto (5 Bytes)
??:??:?? 1   module_config.proto.tmp (53 Bytes)
??:??:?? 1   cannedConf.proto (12 Bytes)
??:??:?? 1   config.proto (68 Bytes)
??:??:?? 1   module.proto.tmp (30 Bytes)
??:??:?? 1   module.proto (30 Bytes)
??:??:?? 1   channels.proto (53 Bytes)
??:??:?? 1 I2C device found at address 0x51
??:??:?? 1 PCF8563 RTC found
??:??:?? 1 I2C device found at address 0x77
??:??:?? 1 Wire.available() = 1
??:??:?? 1 BME-280 sensor found at address 0x77
??:??:?? 1 2 I2C devices found
??:??:?? 1 Meshtastic hwvendor=7, swver=1.3.25.85f46d3
??:??:?? 1 Reset reason: 0x1
??:??:?? 1 Setting random seed 2053277982
??:??:?? 1 Initializing NodeDB
??:??:?? 1 No /prefs/db.proto preferences found
??:??:?? 1 Installing default DeviceState
??:??:?? 1 Initial packet id 1159555873, numPacketId 4294967295
??:??:?? 1 Loading /prefs/config.proto
??:??:?? 1 Loaded saved config version 13
??:??:?? 2 Loading /prefs/module.proto
??:??:?? 2 Loaded saved moduleConfig version 13
??:??:?? 2 Loading /prefs/channels.proto
??:??:?? 2 Loaded saved channelFile version 13
??:??:?? 2 Expanding short PSK #1
??:??:?? 2 Wanted region 3, using EU868
??:??:?? 2 region=3, NODENUM=0x6fce0454, dbsize=1
??:??:?? 2 Can't write prefs
??:??:?? 2 Can't write prefs
??:??:?? 2 Saving /prefs/module.proto
??:??:?? 2 assert failed /home/runner/.platformio/packages/framework-arduinoadafruitnrf52/libraries/Adafruit_LittleFS/src/littlefs/lfs.c: ����������������������������������������������%
```

garthvh, 2022-07-27:
Can you try 1.3.27, it fixed a radio based reboot loop and I would like to be able to rule that out.

arnotron, 2022-07-27:
Hello,
unfortunately 1.3.27 does not fix the issue. Nonetheless please find the new console output below:

��H??:??:?? 1

��H??:??:?? 1

//\ E S H T /\ S T / C
q^[[A^M
��H??:??:?? 3

//\ E S H T /\ S T / C

??:??:?? 3 Filesystem files:
??:??:?? 3   radio.proto (5 Bytes)
??:??:?? 3   module_config.proto.tmp (53 Bytes)
??:??:?? 3   cannedConf.proto (12 Bytes)
??:??:?? 3   config.proto (68 Bytes)
??:??:?? 3   module.proto.tmp (30 Bytes)
??:??:?? 3   module.proto (30 Bytes)
??:??:?? 3   channels.proto (53 Bytes)
??:??:?? 3 I2C device found at address 0x51
??:??:?? 3 PCF8563 RTC found
??:??:?? 3 I2C device found at address 0x77
??:??:?? 3 Wire.available() = 1
??:??:?? 3 BME-280 sensor found at address 0x77
??:??:?? 3 2 I2C devices found
??:??:?? 3 Meshtastic hwvendor=7, swver=1.3.27.c88ba58
??:??:?? 3 Reset reason: 0x1
??:??:?? 3 Setting random seed 251135736
??:??:?? 3 Initializing NodeDB
??:??:?? 3 No /prefs/db.proto preferences found
??:??:?? 3 Installing default DeviceState
??:??:?? 3 Initial packet id 19921269, numPacketId 4294967295
??:??:?? 3 Loading /prefs/config.proto
??:??:?? 3 Loaded saved config version 13
??:??:?? 3 Loading /prefs/module.proto
??:??:?? 3 Loaded saved moduleConfig version 13
??:??:?? 4 Loading /prefs/channels.proto
??:??:?? 4 Loaded saved channelFile version 13
??:??:?? 4 Expanding short PSK #1
??:??:?? 4 Wanted region 3, using EU868
??:??:?? 4 region=3, NODENUM=0x6fce0454, dbsize=1
??:??:?? 4 Can't write prefs
??:??:?? 4 Can't write prefs
??:??:?? 4 Saving /prefs/module.proto
??:??:?? 4 assert failed /home/runner/.platformio/packages/framework-arduinoadafruitnrf52/libraries/Adafruit_LittleFS/src/littlefs/lfs.c: ����������������������������������������������%

arnotron, 2022-08-02:
Just for clarification: I am not sure if my problem is related to the specific sensor I tried to enable. The console log sounds a bit like the configuration is corrupted. If there is some further troubleshooting I could do or load a debugging version of the firmware I’d be willing to do so.

arnotron, 2022-08-02:
Closed by mistake. Just wanted to provide an update.

arnotron, 2022-08-03:
Hello,
thank you very much, the filesystem reset did help get my device back in working condition. I could also enable the BME280 sensor without bringing it into the same condition.

I changed the subject so this is clearer. As you said you are already chasing it down I think this Bug can be closed.

Kind regards, Arne

GoldProto, 2022-08-06:
Thanks for the pointer lyusopov. It did not help my devices to work. 

Please see the issue noted above by lyusopov for latest updates on my findings. 
Release  (Meshtastic) 1.3.31.0084643 did not help either. 
Thank you.

thebentern, 2022-10-17:
Not seeing this anymore. If it occurs again on the latest firmware, please file a new issue.

closed 2022-10-17 by thebentern

marek22k, 2023-12-28:
I had a similar error (also with T-Echo with sensor) and a factory erase helped:
```
Welcome to minicom 2.9

OPTIONS: I18n 
Compiled on Sep 23 2023, 19:55:08.
Port /dev/ttyACM0, 17:13:09

Press CTRL-A Z for help on special keys

斆@INFO  | ??:??:?? 2 

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 2 Filesystem files:
DEBUG | ??:??:?? 2  module.proto (100 Bytes)
DEBUG | ??:??:?? 2  channels.proto (143 Bytes)
DEBUG | ??:??:?? 2  neighbors.proto (42 Bytes)
DEBUG | ??:??:?? 2  db.proto (363 Bytes)
DEBUG | ??:??:?? 2 Using analog input 4 for battery level
INFO  | ??:??:?? 2 Scanning for i2c devices...
DEBUG | ??:??:?? 2 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 2 I2C device found at address 0x51
INFO  | ??:??:?? 2 PCF8563 RTC found
DEBUG | ??:??:?? 2 I2C device found at address 0x77
[23;80H [24;1HDEBUG | ??:??:?? 2 Wire.available() = 1
[23;80H [24;1HINFO  | ??:??:?? 2 BME-280 sensor found at address 0x77
[23;80H [24;1HINFO  | ??:??:?? 2 2 I2C devices found
[23;80H [24;1HDEBUG | ??:??:?? 2 acc_info = 0
[23;80H [24;1HDEBUG | ??:??:?? 2 found i2c sensor meshtastic_TelemetrySensorType_BME280
[23;80H [24;1HINFO  | ??:??:?? 2 Meshtastic hwvendor=7, swver=2.2.16.1c6acfd
[23;80H [24;1HDEBUG | ??:??:?? 2 Reset reason: 0x0
[23;80H [24;1HDEBUG | ??:??:?? 2 Setting random seed 3521160043
[23;80H [24;1HINFO  | ??:??:?? 2 Initializing NodeDB
[23;80H [24;1HINFO  | ??:??:?? 2 Loading /prefs/db.proto
[23;80H [24;1HERROR | ??:??:?? 2 assert failed /home/runner/.platformio/packages/framework-arduinoadafruitnrf52/libraries/Adafruit_LittleFS/src
```
In my case, the error occurred after I clicked on "Shutdown" in the Android app. After that, the device no longer started and this error appeared in the serial console. How can I prevent this from happening in the future?