meshtastic/firmware #1568: [1.3 Bug]: Doing a get request causes device to reboot

garthvh, 2022-07-26:
### Category

Other

### Hardware

T-Beam

### Firmware Version

1.3.27

### Description

When trying to do an admin message get request for getCannedMessageModulePart1Request the tbeam reboots

### Relevant log output

```Shell
??:??:?? 93 To Radio onwrite
??:??:?? 93 PACKET FROM PHONE (id=0x5a1a2f67 Fr0x00 To0xf0, WantAck1, HopLim0 Ch0x0 Portnum=6 priority=70)
??:??:?? 93 Enqueued local (id=0x5a1a2f67 Fr0x00 To0xf0, WantAck1, HopLim0 Ch0x0 Portnum=6 priority=70)
??:??:?? 93 [Router] Add packet record (id=0x5a1a2f67 Fr0x00 To0xf0, WantAck1, HopLim0 Ch0x0 Portnum=6 priority=70)
??:??:?? 93 [Router] handleReceived(REMOTE) (id=0x5a1a2f67 Fr0x00 To0xf0, WantAck1, HopLim0 Ch0x0 Portnum=6 priority=70)
??:??:?? 93 [Router] Module 'Admin' wantsPacket=1
??:??:?? 93 [Router] Received Admin from=0x0, id=0x5a1a2f67, portnum=6, payloadlen=3
??:??:?? 93 [Router] Client is getting radio canned message part1
??:??:?? 93 [Router] *** handleGetCannedMessageModulePart1
assertion "req.decoded.want_response" failed: file "src/modules/CannedMessageModule.cpp", line 376, function: void CannedMessageModule::handleGetCannedMessageModulePart1(const MeshPacket&, AdminMessage*)
abort() was called at PC 0x4012d85b on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008fe3f:0x3ffd5bd0 0x400901c5:0x3ffd5bf0 0x4012d85b:0x3ffd5c10 0x400edc3d:0x3ffd5c40 0x400ede1d:0x3ffd5c60 0x400dfc02:0x3ffd5c80 0x400ed710:0x3ffd5cb0 0x400ecd5e:0x3ffd5db0 0x400dfb8e:0x3ffd5ec0 0x400e4159:0x3ffd5ef0 0x400e4211:0x3ffd5f10 0x400e4231:0x3ffd5f30 0x400e39b5:0x3ffd5f60 0x400d557e:0x3ffd5f80 0x400f6c95:0x3ffd5fa0 0x400de074:0x3ffd5fd0 0x4012c07d:0x3ffd5ff0

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
WARNING file:mesh_interface.py _sendToRadio line:467 Not sending packet because protocol use is disabled by noProto
??:??:?? 0
```

caveman99, 2022-07-28:
The assert that is bombing asks for this to be set to true ... 

```
  /*
   * Not normally used, but for testing a sender can request that recipient
   * responds in kind (i.e. if it received a position, it should unicast back it's position).
   * Note: that if you set this on a broadcast you will receive many replies.
   */
  bool want_response = 3;
```

a) does the request set this to TRUE? I assume not, hence the bomb.
b) does anybody know why this has to be set to TRUE? from the comment looks like a bug, since this is not intended to go over radio, this is wire/admin api only. It should reply with the chosen dataset anyway, regardless of this flag. If so, we can simply remove the 4 asserts

garthvh, 2022-07-31:
I will double check my code, but I always set want response to true

garthvh, 2022-07-31:
https://github.com/meshtastic/Meshtastic-Apple/blob/a06debe968800732dcfa0cd13e164db4bd3a8a59/Meshtastic/Helpers/BLEManager.swift#L1143

caveman99, 2022-07-31:
i'm not sure this is the same thing. the assert specifically looks for a field named want_response in the decoded payload part of meshpacket (message "Data"). want_ack is in the plaintext header (message "MeshPacket").

Again I have no idea what this assert is for ...

garthvh, 2022-08-02:
I did both and each individual, still crashes

		meshPacket.wantAck = wantResponse
		meshPacket.decoded.wantResponse = wantResponse

garthvh, 2022-08-03:
found the right want response datamessage.wantresponse

closed 2022-08-03 by garthvh

caveman99, 2022-08-03:
still intrigues me why this is neccessary... the assert just makes sure the field is set.