meshtastic/firmware #1569: [Bug]:

pcopa, 2022-07-26:
### Category

Hardware Compatibility

### Hardware

T-Beam

### Firmware Version

1.3.25.85f46d3

### Description

HARDWARE: T-Beam v1.1 w/OLED
OS: Windows 10 64bit
Python 3.10.5 (tags/v3.10.5:f377153, Jun  6 2022, 16:14:13) [MSC v.1929 64 bit (AMD64)] on win32

Please have mercy on me. I'm new to Meshtastic and GitHub, but have managed to clown and install the different firmware's with CLI.

FYI:
The firmware Version": "1.3.25.85f46d3 works best for me at the moment because I can see the stream on COM4 and the collected information from the other device (1.2.63) looks valid.

However firmware-1.3.26.0010231 seems to be locked up in a dead-lock-loop while booting. No errors while uploading. Screen keeps blinking.

### Relevant log output

```Shell
PS D:\Meshtastic\GitHub\firmware-1.3.25.85f46d3> meshtastic.exe --support

If having issues with meshtastic cli or python library
or wish to make feature requests, visit:
https://github.com/meshtastic/Meshtastic-python/issues
When adding an issue, be sure to include the following info:
 System: Windows
   Platform: Windows-10-10.0.22000-SP0
   Release: 10
   Machine: AMD64
   Encoding (stdin): utf-8
   Encoding (stdout): utf-8
 meshtastic: v1.3a20
 Executable: C:\Users\Schelto\AppData\Local\Programs\Python\Python310\Scripts\meshtastic
 Python: 3.10.5 CPython MSC v.1929 64 bit (AMD64)

Please add the output from the command: meshtastic --info

PS D:\Meshtastic\GitHub\firmware-1.3.25.85f46d3> meshtastic.exe --info
Connected to radio

Owner: Mieke (Mie)
My info: { "myNodeNum": 4145900520, "hasGps": true, "firmwareVersion": "1.3.25.85f46d3", "rebootCount": 3, "bitrate": 74.341286, "messageTimeoutMsec": 300000, "minAppVersion": 20300, "maxChannels": 8, "hasWifi": true, "airUtilTx": 0.22180556 }
Nodes in mesh:  {'num': 4145900520, 'user': {'id': '!f71d6be8', 'longName': 'Mieke', 'shortName': 'Mie', 'macaddr': '30:c6:f7:1d:6b:e8', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': xxxxx7436, 'longitudeI': xxxxx7550, 'altitude': xxx, 'time': 1658876078, 'locationSource': 'LOCSRC_GPS_INTERNAL', 'posTimestamp': 1658876023, 'altitudeHae': xxx49911, 'altGeoidSep': xxxx9639, 'PDOP': 198, 'HDOP': 141, 'groundTrack': xxxxx000, 'fixQuality': 1, 'satsInView': 6, 'latitude': xxxxx599999996, 'longitude': xxxxx775499999999}, 'lastHeard': 1658876078, 'deviceMetrics': {'batteryLevel': 100, 'voltage': 4.142, 'airUtilTx': 0.24730556}}

Preferences: { "device": { "ntpServer": "0.pool.ntp.org" }, "position": { "positionFlags": 3 }, "power": { "lsSecs": 300 }, "wifi": {}, "display": {}, "lora": {} }

Module preferences: { "mqtt": {}, "serial": {}, "externalNotification": {}, "rangeTest": {}, "telemetry": {}, "cannedMessage": {} }

Channels:
  PRIMARY psk=default { "psk": "AQ==" }

Primary channel URL: https://www.meshtastic.org/e/#CgMiAQE  ( is link correct??)

PS D:\Meshtastic\GitHub\firmware-1.3.25.85f46d3>
```

mc-hamster, 2022-07-27:
> However firmware-1.3.26.0010231 seems to be locked up in a dead-lock-loop while booting. No errors while uploading. Screen keeps blinking.

Yeah ... that release was unstable. We yanked it. 1.3.27 is where it's at now.

If you upgrade, please keep your nodes on the same version during the preview phase.

pcopa, 2022-07-27:
Thanks @mc-hamster 
1.3.27 is uploaded to both devices and is running.

closed 2022-07-27 by pcopa