meshtastic/firmware #1590: 1.3 Testing Punchlist

garthvh, 2022-08-03:
## Config Sections

* [ ] Device Config
* [ ] Display Config
* [ ] LoRa Config
* [ ] Power Config
* [ ] Position Config
* [ ] WiFi Config

Waffle2, 2022-08-04:
Testing on 1.3.29.7afc149, with tlora-v2-1-1.6

Configuring from command line with meshtastic:
`meshtastic --set position.fixed_position true
Connected to radio
Set position.fixed_position to true
Writing modified preferences to device
$ meshtastic --setlat 37.0 --setlon -121.0 
Connected to radio
Aborting due to: 'LocalConfig' object has no attribute 'fixed_position'`

meshtastic --set wifi.ap_mode false
Connected to radio
Set wifi.ap_mode to false
Writing modified preferences to device

$ meshtastic --set wifi.ssid "Localnet"
Connected to radio
Set wifi.ssid to Localnet
Writing modified preferences to device

$ meshtastic --set wifi.password SecretPass47
Connected to radio
Aborting due to: local variable 'localConfig' referenced before assignment

There are several more, very similar to this.  Attempting to configure from the IOS app crashes the IOS app as well.

garthvh, 2022-08-04:
the right command for wifi password is set wifi.psk "password"

garthvh, 2022-08-04:
When you see the localconfig errors the command is wrong Aborting due to: 'LocalConfig' object has no attribute

garthvh, 2022-08-04:
> Aborting due to: 'LocalConfig' object has no attribute 'fixed_position'

Only Config Sections and Module Config are working for setting in my testing, I would not expect any of the convenience methods that are setting things to work.

sachaw, 2022-08-04:
I think there is a case for adding an enabled Boolean property to the wifi config.

Waffle2, 2022-08-04:
> the right command for wifi password is set wifi.psk "password"

Ok, thanks - I was using https://meshtastic.org/docs/settings/config/wifi as my reference.

garthvh, 2022-08-04:
> > the right command for wifi password is set wifi.psk "password"
> 
> Ok, thanks - I was using https://meshtastic.org/docs/settings/config/wifi as my reference.

Thanks! Fixed https://github.com/meshtastic/Meshtastic/pull/386

Waffle2, 2022-08-04:
Configuring via the web interface (via wifi) fails when attempting to save changes in the User configuration panel.  Changes are rejected with a status message "hwModel must be a valid enum value" in red by the (unchangeable) Hardware field.  In my case, the field shows TLORA_V2_1_1p6

sachaw, 2022-08-06:
> Configuring via the web interface (via wifi) fails when attempting to save changes in the User configuration panel. Changes are rejected with a status message "hwModel must be a valid enum value" in red by the (unchangeable) Hardware field. In my case, the field shows TLORA_V2_1_1p6

This is now fixed

GUVWAF, 2022-10-13:
In my testing on 1.3.44 both --sendping and --reply are already working. 
When using 'meshtastic --reply' on one device, sending a message 'Hi there' from another gets a response "got msg 'Hi there' with rxSnr: 11.0 and hopLimit: 3" as seen here: 

> 17:46:49 300 [RadioIf] AirTime - Packet received : 1214ms
17:46:49 300 [Router] Add packet record (id=0x88ec4132 Fr0xe4 To0xff, WantAck0, HopLim3 Ch0x8 encrypted rxSNR=10.25 rxRSSI=1.84093)
17:46:49 301 [Router] Using channel 0 (hash 0x8)
17:46:49 301 [Router] Expanding short PSK #1
17:46:49 301 [Router] Using AES128 key!
17:46:49 301 [Router] ESP32 crypt fr=bff18ce4, num=88ec4132, numBytes=55!
17:46:49 301 [Router] decoded message (id=0x88ec4132 Fr0xe4 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=1 rxtime=1665683209 rxSNR=10.25 rxRSSI=1.84093)
17:46:49 301 [Router] handleReceived(REMOTE) (id=0x88ec4132 Fr0xe4 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=1 rxtime=1665683209 rxSNR=10.25 rxRSSI=1.84093)
17:46:49 301 [Router] Module 'text' wantsPacket=1
17:46:49 301 [Router] Received text msg from=0xbff18ce4, id=0x88ec4132, msg=got msg 'Hi there' with rxSnr: 11.0 and hopLimit: 3

thebentern, 2022-10-13:
> In my testing on 1.3.44 both --sendping and --reply are already working. When using 'meshtastic --reply' on one device, sending a message 'Hi there' from another gets a response "got msg 'Hi there' [...]

Checked off! :-)

geeksville, 2022-10-14:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/all-ttgo-boards-but-one-are-rebooting-after-each-cli-command/6600/2

garthvh, 2022-10-21:
Closing this list and making issues for the remaining 4 items

closed 2022-10-21 by garthvh