meshtastic/firmware #1595: Add an enabled Boolean property to the wifi config.

garthvh, 2022-08-04:
I think there is a case for adding an enabled Boolean property to the wifi config.

_Originally posted by @sachaw in https://github.com/meshtastic/Meshtastic-device/issues/1590#issuecomment-1204808327_

garthvh, 2022-08-04:
Since there is now field length validation for ssid and psk and the old method of setting them to "" no longer works, we need an enabled boolean so clients can disable wifi without flashing firmware

Noki, 2022-08-04:
I think it would be a good idea to consolidate this, ap_mode and ap_hidden into a single ENUM:

wifi.mode (`off`,`client`,`ap`,`ap_hidden`)

In my opinion this would be a lot easier to understand for most users.

garthvh, 2022-08-05:
Complete

caveman99, 2022-08-05:
this is only the settings for now, please leave ths open until the setting is wired to the firmware :-) I was planning on doing that after the weekend.

mc-hamster, 2022-08-05:
Updated title to represent the implementation.

sachaw, 2022-08-06:
As @Noki said, I too believe this is the way forwards https://github.com/meshtastic/Meshtastic-device/issues/1601

garthvh, 2022-08-06:
Enabled is not part of an enum anywhere else are we going to make this change to all 6 modules with enabled? They does not match our other config at all.  Has anyone mocked up a UI with this structure, seems confusing.

sachaw, 2022-08-07:
Enum is now implemented, we still require the logic to be implemented to support the disabled property

PR #1677 (2022-09-06): Wifi enabled plumbed in
Closes #1595
closed 2022-09-06 by thebentern