meshtastic/firmware #1598: [Bug]: 1.3 ESP32 NimBLE API hangs

andrekir, 2022-08-05:
### Category

BLE

### Hardware

T-Beam

### Firmware Version

1.3

### Description

ESP32 Bluetooth connection hangs using new NimBLE API. same behavior confirmed on Android and iOS:

- connection is present
- phones stop receiving data after NimBLE hangs (users can send but not receive data)

switching back to the previous Bluetooth stack works (removing `-DUSE_NEW_ESP32_BLUETOOTH` from `platformio.ini`).

garthvh, 2022-08-05:
The classic bluetooth stack is not a fix for apple os's, SwiftUI does not support classic bluetooth at all and 1.2 devices are crashing on average once every session, so we need to get the new stack working correctly, reverting to our old stack is not a solution for the apple apps.

garthvh, 2022-08-05:
These are all BLE 4.0 specific errors in my testing, Invalid Handle and The request is not supported are the errors I am getting from tbeams mostly.

thebentern, 2022-08-05:
I have a PR up for updating our NimBLE lib_dep version. We're quite a bit behind. I have no idea if it'll fix this issue or not, but it's worth a shot.

diamondbarcode, 2022-08-09:
This happen to TLora too

garthvh, 2022-08-09:
Log from discord

MESHTASTIC MESH ACTIVITY LOG
✅ BLE Connecting: Meshtastic_e214 - 8/8/22 5:10:34.8420 
✅ BLE Connected: Meshtastic_e214 - 8/8/22 5:10:35.1760 
✅ BLE Service for Meshtastic discovered by Meshtastic_e214 - 8/8/22 5:10:35.5240 
✅ BLE Service for Meshtastic discovered by Meshtastic_e214 - 8/8/22 5:10:35.5290 
✅ BLE Service for Meshtastic discovered by Meshtastic_e214 - 8/8/22 5:10:35.5330 
✅ BLE did discover TORADIO characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:35.5850 
✅ BLE did discover FROMRADIO characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:35.5970 
✅ BLE did discover FROMNUM (Notify) characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:35.6040 
✅ BLE did discover TORADIO characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:35.6440 
✅ BLE did discover FROMRADIO characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:35.6490 
✅ BLE did discover FROMNUM (Notify) characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:35.6530 
✅ BLE did discover TORADIO characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:35.7620 
✅ BLE did discover FROMRADIO characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:35.7690 
✅ BLE did discover FROMNUM (Notify) characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:35.7760 
ℹ️ BLE Disconnected: Meshtastic_e214: User Initiated Disconnect - 8/8/22 5:10:53.9410 
✅ BLE Connecting: Meshtastic_e214 - 8/8/22 5:10:55.9940 
✅ BLE Connected: Meshtastic_e214 - 8/8/22 5:10:56.5980 
✅ BLE Service for Meshtastic discovered by Meshtastic_e214 - 8/8/22 5:10:56.9500 
✅ BLE Service for Meshtastic discovered by Meshtastic_e214 - 8/8/22 5:10:56.9560 
✅ BLE Service for Meshtastic discovered by Meshtastic_e214 - 8/8/22 5:10:56.9600 
✅ BLE Service for Meshtastic discovered by Meshtastic_e214 - 8/8/22 5:10:56.9630 
✅ BLE did discover TORADIO characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:57.0100 
✅ BLE did discover FROMRADIO characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:57.0140 
✅ BLE did discover FROMNUM (Notify) characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:57.0200 
✅ BLE did discover TORADIO characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:57.0730 
✅ BLE did discover FROMRADIO characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:57.0790 
✅ BLE did discover FROMNUM (Notify) characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:57.0850 
✅ BLE did discover TORADIO characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:57.1340 
✅ BLE did discover FROMRADIO characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:57.1430 
✅ BLE did discover FROMNUM (Notify) characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:57.1470 
✅ BLE did discover TORADIO characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:57.2510 
✅ BLE did discover FROMRADIO characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:57.2570 
✅ BLE did discover FROMNUM (Notify) characteristic for Meshtastic by Meshtastic_e214 - 8/8/22 5:10:57.2610 
💾 Saved a new myInfo for node number: 3304579604 - 8/8/22 5:10:57.4220 
💾 Updated myInfo for node number: 3304579604 - 8/8/22 5:10:57.6150 
💾 Updated myInfo for node number: 3304579604 - 8/8/22 5:10:57.8500 
💾 Updated myInfo for node number: 3304579604 - 8/8/22 5:10:58.1240 
💾 BLE FROMRADIO received and nodeInfo inserted for Meshtastic e214 - 8/8/22 5:10:58.4300 
💾 BLE FROMRADIO received and nodeInfo inserted for Meshtastic f090 - 8/8/22 5:10:58.5190 
💾 Updated Device Config for node number: 3304579604 - 8/8/22 5:10:58.5280 
🗺️ Custom Position config received 3304579604 - 8/8/22 5:10:58.5950 
💾 Updated Position Config for node number: 3304579604 - 8/8/22 5:10:58.5980 
📶 Default WiFi config received 3304579604 - 8/8/22 5:10:58.6880 
💾 Updated WiFi Config for node number: 3304579604 - 8/8/22 5:10:58.6950 
💾 Updated Display Config for node number: 3304579604 - 8/8/22 5:10:58.7500 
📻 Custom LoRa config 3304579604 - 8/8/22 5:10:58.8080 
💾 Updated LoRa Config for node number: 3304579604 - 8/8/22 5:10:58.8140 
🤖 Default Serial Module config 3304579604 - 8/8/22 5:10:58.9280 
💾 Updated Serial Module Config for node number: 3304579604 - 8/8/22 5:10:58.9460 
💾 Updated External Notification Module Config for node number: 3304579604 - 8/8/22 5:10:58.9970 
💾 Updated Range Test Config for node number: 3304579604 - 8/8/22 5:10:59.1230 
💾 Updated Telemetry Module Config for node number: 3304579604 - 8/8/22 5:10:59.1750 
💾 Updated Canned Message Module Config for node number: 3304579604 - 8/8/22 5:10:59.2370 
🤜 BLE Config Complete Packet Id: 8 - 8/8/22 5:10:59.2890 
💾 Updated Node Position Coordinates, SNR and Time from Position App Packet For: 3304579604 - 8/8/22 5:10:59.4110 
💾 Updated Node Position Coordinates, SNR and Time from Position App Packet For: 3045257360 - 8/8/22 5:10:59.4150 
🕸️ ROUTING PACKET received for RequestID: 4220625949 Error: This message is not a failure - 8/8/22 5:10:59.4680 
💾 Updated Node Position Coordinates, SNR and Time from Position App Packet For: 3045257360 - 8/8/22 5:10:59.5320 
💾 Telemetry Saved for Node: 3045257360 - 8/8/22 5:10:59.5950 
💾 Updated NodeInfo SNR 6.0 and Time from Node Info App Packet For: 3045257360 - 8/8/22 5:10:59.6570 
💾 Updated Node Position Coordinates, SNR and Time from Position App Packet For: 3045257360 - 8/8/22 5:10:59.7240 
[... cut ...]
📍 Sent a Position Packet from the Apple device GPS to node: 3304579604 - 8/8/22 5:46:40.4550 
💾 Telemetry Saved for Node: 3304579604 - 8/8/22 5:46:40.7960 
📲 New messageId 1645745365 sent to Meshtastic f090 - 8/8/22 5:57:59.9650 
💾 Saved a new sent message from 3304579604 to 3045257360 - 8/8/22 5:57:59.9810 
📲 New messageId 640478451 sent to All - Broadcast - 8/8/22 5:58:19.4040 
💾 Saved a new sent message from 3304579604 to 4294967295 - 8/8/22 5:58:19.4160 
📍 Sent a Position Packet from the Apple device GPS to node: 3304579604 - 8/8/22 5:59:45.5840

tropho23, 2022-08-09:
I also experienced this bug; it happens for my T-Beam v1.1, Tlora v2.1-1.6 and Heltec v2.1 boards with 1.3.34 alpha firmware.

diamondbarcode, 2022-08-09:
i test firmware 1.3.31 / android app 1.3.30 .. the 2 Tloralv2.1.1.6 app did not received message at all but can send out however when message send i can see on LCD screen. I reboot phone i got flood with all missing message.

T-Beam problems seem solve - 3h test just reboot 1 time - i observed 2 this and phone see if anything change

ndoolin, 2022-08-10:
I am also having this issue... messages appear on the device screens but have issues getting to their apps on iOS and Android.
1.3.34 fw on t-beam and tlora v2.1.6 devices running current apps on iOS and Android Devices.  Closing the App completely and reconnecting to the device usually helps get the message to the app.

thebentern, 2022-08-14:
Latest attempted fixes are being pushed to this PR (see artifacts)
https://github.com/meshtastic/Meshtastic-device/pull/1629

Enhanced logging is enabled on this build, so serial logs when bugs are encountered are very helpful for debugging.

andrekir, 2022-08-23:
solved.

closed 2022-08-23 by andrekir