meshtastic/firmware #1603: 1.3 Bug - BLE Strange Race Condition Between Nodes and Config on WantConfig Request

garthvh, 2022-08-06:
If you delete the iOS app (which clears all local node database data), forget the device under settings > bluetooth and simulate the device connecting for the first time I am consistently getting all of my config values after I have received my myInfo, but before I have received the nodeInfo packets.

My settings page is set up so that in the future if the admin channel is enabled you can administer any node on the network, but because of this I need to know the node num of the node before I can save the config.

I was just going to run wantConfig again, but when testing this build from @andrekir https://github.com/andrekir/Meshtastic-device/actions/runs/2798003724 I noticed that on first connection the old BLE stack always returns nodeInfo before the config.

garthvh, 2022-08-06:
✅ BLE Connected: Meshtastic_89f0
✅ BLE Service for Meshtastic discovered by Meshtastic_89f0
✅ BLE did discover TORADIO characteristic for Meshtastic by Meshtastic_89f0
✅ BLE did discover FROMRADIO characteristic for Meshtastic by Meshtastic_89f0
✅ BLE did discover FROMNUM (Notify) characteristic for Meshtastic by Meshtastic_89f0
💾 Updated myInfo for node number: 4145973744
💾 Updated Device Config for node number: 4145973744
🗺️ Custom Position config received 4145973744
💾 Updated Position Config for node number: 4145973744
📶 Default WiFi config received 4145973744
💾 Updated WiFi Config for node number: 4145973744
🖥️ Default Display config
💾 Updated Display Config for node number: 4145973744
📻 Custom LoRa config 4145973744
💾 Updated LoRa Config for node number: 4145973744
🤖 Default Serial Module config 4145973744
💾 Updated Serial Module Config for node number: 4145973744
🚨 Default External Notifiation Module config
💾 Updated External Notification Module Config for node number: 4145973744
⛰️ Default Range Test Module config
💾 Updated Range Test Config for node number: 4145973744
📈 Default Telemetry Module config
💾 Updated Telemetry Module Config for node number: 4145973744
🥫 Default Canned Message Module config
💾 Updated Canned Message Module Config for node number: 4145973744
💾 BLE FROMRADIO received and nodeInfo inserted for Meshtastic 89f0
💾 BLE FROMRADIO received and nodeInfo inserted for Meshtastic 0e84
🤜 BLE Config Complete Packet Id: 18

ernax78, 2022-08-07:
Also on  "firmwareVersion": "1.3.32.7e6c22f",  'hwModel': 'TLORA_V2'

garthvh, 2022-08-07:
https://github.com/meshtastic/Meshtastic-device/pull/1606

closed 2022-08-07 by garthvh

garthvh, 2022-08-07:
Perfect

✅ BLE Connecting: Meshtastic_89f0
✅ BLE Connected: Meshtastic_89f0
✅ BLE Service for Meshtastic discovered by Meshtastic_89f0
✅ BLE did discover TORADIO characteristic for Meshtastic by Meshtastic_89f0
✅ BLE did discover FROMRADIO characteristic for Meshtastic by Meshtastic_89f0
✅ BLE did discover FROMNUM (Notify) characteristic for Meshtastic by Meshtastic_89f0
💾 Saved a new myInfo for node number: 4145973744
💾 Saved the All - Broadcast User
💾 BLE FROMRADIO received and nodeInfo inserted for Meshtastic 89f0
💾 BLE FROMRADIO received and nodeInfo inserted for Meshtastic 0e84
📟 Custom Device config
💾 Updated Device Config for node number: 4145973744
🗺️ Custom Position config received 4145973744
💾 Updated Position Config for node number: 4145973744
📶 Default WiFi config received 4145973744
💾 Updated WiFi Config for node number: 4145973744
🖥️ Default Display config
💾 Updated Display Config for node number: 4145973744
📻 Default LoRa config 4145973744
💾 Updated LoRa Config for node number: 4145973744
🤖 Default Serial Module config 4145973744
💾 Updated Serial Module Config for node number: 4145973744
🚨 Default External Notifiation Module config
💾 Updated External Notification Module Config for node number: 4145973744
⛰️ Default Range Test Module config
💾 Updated Range Test Config for node number: 4145973744
📈 Default Telemetry Module config
💾 Updated Telemetry Module Config for node number: 4145973744
🥫 Default Canned Message Module config
💾 Updated Canned Message Module Config for node number: 4145973744
🤜 BLE Config Complete Packet Id: 2