meshtastic/firmware #1610: 1.3 [Bug] New Routing with two units delivers message but sometimes the ACK returned is Max Retransmission Reached

garthvh, 2022-08-08:
Seems to only happen when there are only two devices on the network.

In the image below all of the messages were received by the other device and shows on the device screen

[image]

GUVWAF, 2022-08-09:
This might not be a bug, but something that may happen in an RF environment. If the implicit ACK somehow does not end up at the sender, it will do a retransmission. However, then the other node will not send an implicit ACK again, because it already did. I see in the source that only packets with want_ack will ACK a resent packet. 
The reason why you only see this with two nodes in the network could be that then another node’s implicit ACK will end up at the sender.

GUVWAF, 2022-08-11:
feh123 reported this on Discord as well. 
We could actually let nodes re-ACK a retransmission for broadcast messages as well, but we should be careful that we are not ACK-ing a rebroadcast again, as this would overload the channel. Right now, the receiving node will likely log “_Ignoring incoming msg, because we've already seen it_” when it receives the retransmission. This is good behavior for _rebroadcasts_, but maybe not for _retransmissions_. In order to re-ACK a retransmission, we could check whether the hop_limit is still at maximum, so we know it is the original sender that is retransmitting. 
I would like to know whether this is the desired behavior, then I could try to implement and test this.

garthvh, 2022-08-12:
Seems especially pronounced in the same room while testing, I have not been able to replicate this very well with more distance between the nodes.

garthvh, 2022-08-12:
> 

This sounds right to me.

closed 2022-08-13 by garthvh