meshtastic/firmware #1612: [Bug]:

nomore1007, 2022-08-08:
### Category

WiFi

### Hardware

T-Beam

### Firmware Version

1.3.34

### Description

New update wiped wifi info out, and will not accept new wifi settings from python.

garthvh, 2022-08-08:
Settings are indeed wiped out when upgrading from 1.2 to 1.3

The syntax has changed for the new CLI 

meshtastic --set wifi.psk "yourpassword"
meshtastic --set wifi.ssid "ssid"
meshtastic --set wifi.enabled true

nomore1007, 2022-08-08:
set the ssid and psk via this method, but it is not retaining it. I use --info

nomore1007, 2022-08-08:
wifi.enabled true is for the softAP, correct?

garthvh, 2022-08-08:
> I set both. I did not set enabled, that's if you want the softAP? I am connecting it to a wifi network, after I set it, I check with --get, and its still blank.
> 
> Also, I upgraded from 1.3.30 to 1.3.34, that's when it disappeared.

No enabled has been added to wifi in general because the new validation on ssid and psk prevents the old method to disable wifi of setting them to "" the mode is now an enum that defaults to client so you should not need to set it unless you want to turn on the soft ap.

nomore1007, 2022-08-08:
Ok I did leave it unset. I noticed that it also lost the Name and defaulted back to "Meshtastic ####". I downgraded back to .30 and its working again.

nomore1007, 2022-08-08:
```
meshtastic --port /dev/ttyACM1 --set wifi.enabled true
Connected to radio
Aborting due to: local variable 'localConfig' referenced before assignment
```

garthvh, 2022-08-09:
It is updated on the device, looks like the protobufs need to be updated on the python API.  The new settings are in iOS.

closed 2022-08-09 by garthvh