meshtastic/firmware #1619: [Board]: LoraType

smartua7, 2022-08-11:
### SOC

ESP32

### Lora IC

 E22-900M22S (SX1262)

### Product Link

At the moment we are creating a website, very intensively

### Description

Hello, we are reaearch & development laboratory in Ukraine. (Automation.art). We have created a small communication device called LoraType. We created this device for open source and people. It uses radio signal on free frequency, which allows users to be independent of Internet connection, what is crucial in case of any armageddon situation.
We used a very good lora module. Our idea is to create a mesh network of devices. The devices can be used to communicate with relatives and between people as a communicator of last resort in disasters such as war or weather disasters.
Now we are creating a website with complete information and we are also looking for Crowdfunding and of course it is totally free of charge, Loratype has been created not by a Corporation for population, but by people for people. 
We have created a small batch and are currently testing the firmware, we would like to request Meshtastic support for our device or interoperability assistance.
I add pictures for you.

[image]
[image]
Sincerely, we wish you a peaceful sky!

garthvh, 2022-08-11:
Awesome, how can we help you get meshtastic on here?

smartua7, 2022-08-11:
> Awesome, how can we help you get meshtastic on here?

Thank you for your reply!  We are interested in physical tests.  Has the mesh network been tested on 50+ or ​​100+ devices?  Do you have such data on real testing?  and how much can you trust the virtual simulation of the algorithm?

garthvh, 2022-08-12:
The simulator here is very good, you want to use 1.3 as 1.2 is frozen and not getting any changes at this point https://github.com/GUVWAF/Meshtasticator

We have a ham network that runs with 40-50 nodes that I think is the largest running network, the limit in our new 1.3 release will be 64-80 nodes on a single network.

smartua7, 2022-10-07:
Hello, we continue, even due to circumstances in my city.
Please tell me where can I get the source code so that they contain only the functions of a mesh network for integration to our device?

sachaw, 2022-11-20:
The source for the firmware is located in this repo, you can see how other boards have been added by following this tutorial https://meshtastic.org/docs/development/firmware/build#adding-custom-hardware

sachaw, 2023-01-26:
@aelmaker Have you got any update on this? If you are having trouble, there are a number of people that can answer your questions.

garthvh, 2023-05-17:
How do we get the hardware? Can you order it mostly assembled anywhere? A matrix keypad is getting worked on soon, so this may be a target we can support soon.

thebentern, 2023-05-23:
Closing until we hear anything

aelmaker, 2023-06-25:
Helo all! 
Now we have a filled git with information about the device: https://github.com/AutomationArt/LoraType and videos of its use, also posted at https://fosstodon.org/@loratype (not advertising). We have a small batch of ready-made devices for followers and testers. And even a presentable box as a bonus)

My question is simple, can you suggest how you would see the integration of meshtastic into our device?
Your vision is interesting, since you have more experience integrating meshtastic into devices.

The main components are:
**Display** (1.54 inch e-paper display module partial refresh E-ink screen, GDEH0154D67)
**Main controller** The ESP32
**Lora Module** E22 series is the first wireless module in China to adopt Semtech's SX1262 RF chip. This chip provides a variety of modulation methods, such as LoRaTM and traditional GFSK. 
**Keyboard** TCA8418RTWR Great built-in simultaneous click capabilities.
**ARGB** Led WS2912b-2020 is an intelligent control LED light source

Latest video of the device: https://www.youtube.com/watch?v=M7Vq4bjurBg

The last photos of the device yesterday)
[image]
[image]
[image]

thebentern, 2023-06-25:
> Helo all! Now we have a filled git with information about the device: https://github.com/AutomationArt/LoraType and videos of its use, also posted at https://fosstodon.org/@loratype (not [...]

Hi @aelmaker, 

Based on the available hardware, I believe we can produce a firmware variant for this board. After reviewing the schematic, the only thing that would be _new_ to Meshtastic is mapping out the TCA8418 configuration to our input broker. This shouldn't be too difficult though, because the adafruit libraries have this path well-paved. 

Would you be willing to ship out a test device or two to our firmware team for development?

thebentern, 2023-06-27:
> Thank you for your response! Of course! Give us the address! We'll send a few pieces! Always in touch for all the details and help

Awesome! If you don't mind, hit me up via an email at thebentern@meshtastic.org, and I'll send you my address. I'm not too keen on posting it on Github. 😆 
I'll get one of them shipped to @caveman99 as well, who is one of our principal firmware devs.

garthvh, 2023-08-14:
What remains for this?

aelmaker, 2023-08-14:
Hi. We've sent a few devices each to @caveman99 as well as @thebentern 
All in progress!

aelmaker, 2023-10-30:
Hello @caveman99 & @thebentern! Have you received the devices? What are our next steps, what can you say about this for the integration of meshtastic

thebentern, 2023-11-01:
> Hello @caveman99 & @thebentern! Have you received the devices? What are our next steps, what can you say about this for the integration of meshtastic

Howdy! Yes we have received the devices. They look awesome, by the way! I love the silkscreen on the 3D printed case; very nice touch.

I've started preliminary work to integrate them. I have the LoRA radio plumbed in and working. Currently trying to get the e-ink display going with our display lib on the second SPI bus. After that is complete, we'll get started on plumbing in the keyboard.
https://github.com/meshtastic/firmware/pull/2919

fifieldt, 2024-09-16:
Any updates on the LoraType? :)

aelmaker, 2024-09-16:
> Any updates on the LoraType? :)

Hi, we’ve sent devices to @caveman99 and @thebentern for integration and collaboration, but haven’t received any response. Our CEO is attending major LoRaType events (for example, in Germany), and we’re also working on some hardware updates. Regarding MeshTastic, unfortunately, there have been no changes, but overall, we’re staying afloat and selling devices to local followers and for their experiments.

[image]

markbirss, 2024-09-16:
> sent

@thebentern sent me one quite a while, to see if i could get meshtastic working, but the buttons were broken (maybe from shipping) and shorting the pads i could still not get the device to flash

aelmaker, 2024-09-16:
> @aelmaker @fifieldt
> 
> > sent
> 
> @thebentern sent me one quite a while, to see if i could get meshtastic working, but the buttons were broken (maybe from shipping) and shorting the pads i could still not get the device to flash

Hi. It doesn't have a battery (due to the delivery). But it turns on if you supply power to it via USB via TYPE_C. Programming takes place using the POGO PIN CLIP, on the board contacts (if you have an assembled device, you will need to disassemble it), and on the RX TX board. This is also done to make the device accessible to everyone.

fifieldt, 2025-04-20:
@markbirss , any updates?

markbirss, 2025-04-21:
> [@markbirss](https://github.com/markbirss) , any updates?

No, unfortunately not been able to flash the board i have. I will see about how i can make a similiar prototype  as i have such a GDEH0154D67, sx1262 and esp32 around here somewhere

fifieldt, 2025-06-16:
@caveman99 @thebentern @markbirss need any help? I can give it a go if there's a board around :)

markbirss, 2025-08-05:
I have re-started here https://github.com/meshtastic/firmware/pull/7556