meshtastic/firmware #1620: [Bug]: 1.3 BME280 support broken

llamaonaskateboard, 2022-08-11:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

1.3.36.7e03019

### Description

https://github.com/meshtastic/Meshtastic-device/pull/1581 which added support for BMP280 has broken support for BME280 due to a small typo in BMP280Sensor.cpp - https://github.com/meshtastic/Meshtastic-device/blob/v1.3.35.3251cd5/src/modules/Telemetry/Sensor/BMP280Sensor.cpp#L9

### Relevant log output

```Shell
??:??:?? 0 BME-280 sensor found at address 0x76
...
??:??:?? 1 [EnvironmentTelemetryModule] Init sensor: BMP280
??:??:?? 1 [EnvironmentTelemetryModule] Could not connect to detected BMP280 sensor.
 Removing from nodeTelemetrySensorsMap.
```

garthvh, 2022-08-13:
https://github.com/meshtastic/Meshtastic-device/pull/1627

thebentern, 2022-08-13:
Doh! Great catch, @llamaonaskateboard! 
Thanks for hunting down the bug. Fix has been merged.

closed 2022-08-13 by garthvh