meshtastic/firmware #1630: 1.3.37 - Bug No GPS Module on TBeams

garthvh, 2022-08-14:
Looks like we may have turned down the GPS interrogation too much, I am consistently getting No GPS Module on my team with 1.3.37

This is all I see for GPS in the serial logs when doing a reset

??:??:?? 1 WANT GPS=1
??:??:?? 1 GxGSA NOT available

tropho23, 2022-08-14:
Same for me; after installing 1.3.37 I also see a 'No GPS' message on my Tbeam 1.1 6M.  I'm using a Circocom 25mm GPS antenna but that ain't helping me here.

ndoolin, 2022-08-14:
This is from 1.3.32 Firmware so it looks like it may go back further than 1.3.37

//\ E S H T /\ S T / C

??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=reset
??:??:?? 0 Filesystem files:
??:??:?? 0   /prefs/channels.proto (53 Bytes)
??:??:?? 0   /prefs/config.proto (32 Bytes)
??:??:?? 0   /prefs/db.proto (143 Bytes)
??:??:?? 0   /prefs/module.proto (16 Bytes)
??:??:?? 0   /static/Logo_Black.svg.gz (592 Bytes)
??:??:?? 0   /static/icon.svg.gz (842 Bytes)
??:??:?? 0   /static/index.18004c43.js.gz (236277 Bytes)
??:??:?? 0   /static/index.70945424.css.gz (365 Bytes)
??:??:?? 0   /static/index.html.gz (541 Bytes)
??:??:?? 0   /static/robots.txt.gz (42 Bytes)
??:??:?? 0   /static/site.webmanifest.gz (196 Bytes)
??:??:?? 0 I2C device found at address 0x34
??:??:?? 0 axp192 PMU found
??:??:?? 0 I2C device found at address 0x3c
??:??:?? 0 0x7 subtype probed in 2 tries
??:??:?? 0 ssd1306 display found
??:??:?? 0 2 I2C devices found
??:??:?? 0 Meshtastic hwvendor=4, swver=1.3.32.7e6c22f
??:??:?? 0 Setting random seed 3072386082
??:??:?? 0 Total heap: 237792
??:??:?? 0 Free heap: 206036
??:??:?? 0 Total PSRAM: 4194252
??:??:?? 0 Free PSRAM: 4194252
??:??:?? 0 NVS: UsedEntries 81, FreeEntries 549, AllEntries 630, NameSpaces 4
??:??:?? 0 Setup Preferences in Flash Storage
??:??:?? 0 Number of Device Reboots: 3
??:??:?? 0 Initializing NodeDB
??:??:?? 0 Loading /prefs/db.proto
??:??:?? 0 Loaded saved devicestate version 13
??:??:?? 0 Loading /prefs/config.proto
??:??:?? 0 Loaded saved config version 13
??:??:?? 0 Loading /prefs/module.proto
??:??:?? 0 Loaded saved moduleConfig version 13
??:??:?? 0 Loading /prefs/channels.proto
??:??:?? 0 Loaded saved channelFile version 13
??:??:?? 0 Number of Device Reboots: 3
??:??:?? 0 Expanding short PSK #1
??:??:?? 0 Wanted region 0, using Unset
??:??:?? 0 region=0, NODENUM=0x938bfcf0, dbsize=1
??:??:?? 0 Saving /prefs/db.proto
??:??:?? 0 Saving /prefs/config.proto
??:??:?? 0 Saving /prefs/module.proto
??:??:?? 0 Saving /prefs/channels.proto
chip id detect 0x3
Detect CHIP :AXP192
OUTPUT Register 0x5f
??:??:?? 1 AXP192 Begin PASS
??:??:?? 1 DCDC1: ENABLE
??:??:?? 1 DCDC2: ENABLE
??:??:?? 1 LDO2: ENABLE
??:??:?? 1 LDO3: ENABLE
??:??:?? 1 DCDC3: ENABLE
??:??:?? 1 Exten: ENABLE
??:??:?? 1 ----------------------------------------
??:??:?? 1 DCDC1: ENABLE
??:??:?? 1 DCDC2: ENABLE
??:??:?? 1 LDO2: ENABLE
??:??:?? 1 LDO3: ENABLE
??:??:?? 1 DCDC3: ENABLE
??:??:?? 1 Exten: ENABLE
SRC REG:0xc4
Charging enable is enable
Charging target-voltage : 0x2
 end when the charge current is lower than 10% of the set value
Charge current : 700.00 mA
??:??:?? 1 Battery: usbPower=1, isCharging=0, batMv=4187, batPct=100
??:??:?? 1 Read RTC time as 1
??:??:?? 1 Using MSL altitude model
??:??:?? 1 WANT GPS=1
??:??:?? 1 GxGSA NOT available
??:??:?? 1 External Notification Module Disabled
[E][vfs_api.cpp:64] open(): /littlefs/prefs/oem.proto does not exist
??:??:?? 1 No /prefs/oem.proto preferences found
??:??:?? 1 Turning on screen
??:??:?? 1 Module wants a UI Frame
??:??:?? 1 Starting meshradio init...
??:??:?? 1 Set radio: region=Unset, name=LongF, config=0, ch=91, power=30
??:??:?? 1 Radio myRegion->freqStart / myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
??:??:?? 1 Radio myRegion->numChannels: 104
??:??:?? 1 Radio channel_num: 91
??:??:?? 1 Radio frequency: 924.875000
??:??:?? 1 Slot time: 42 msec
??:??:?? 1 Set radio: final power level=20
??:??:?? 1 Current limit set to 100.000000
??:??:?? 1 Current limit set result 0
??:??:?? 1 RF95 init result -19
??:??:?? 1 Frequency set to 924.875000
??:??:?? 1 Bandwidth set to 250.000000
??:??:?? 1 Power output set to 20
??:??:?? 1 Current limit set to 100.000000
??:??:?? 1 Current limit set result 0
??:??:?? 1 RF95 Radio init succeeded, using RF95 radio
??:??:?? 1 Not using WIFI
??:??:?? 1 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=237, time 3188 ms
??:??:?? 1 myNodeInfo.bitrate = 74.341286 bytes / sec
??:??:?? 1 PowerFSM init, USB power=1
??:??:?? 1 Enter state: BOOT
[D][esp32-hal-cpu.c:189] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
??:??:?? 1 [Power] Battery: usbPower=1, isCharging=0, batMv=4187, batPct=100
??:??:?? 1 [Screen] Screen: Started...
??:??:?? 1 [DeviceTelemetryModule] Device Telemetry: Initializing
??:??:?? 1 [DeviceTelemetryModule] -----------------------------------------
??:??:?? 1 [DeviceTelemetryModule] Device Telemetry: Read data
??:??:?? 1 [DeviceTelemetryModule] Telemetry->time: 1
??:??:?? 1 [DeviceTelemetryModule] Telemetry->air_util_tx: 0.000000
??:??:?? 1 [DeviceTelemetryModule] Telemetry->battery_level: 100
??:??:?? 1 [DeviceTelemetryModule] Telemetry->channel_utilization: 0.000000
??:??:?? 1 [DeviceTelemetryModule] Telemetry->voltage: 4

ndoolin, 2022-08-14:
> Same for me; after installing 1.3.37 I also see a 'No GPS' message on my Tbeam 1.1 with NEO-6M GPS chipset. I'm using a Cirocomm 25mm GPS antenna but that ain't helping me here.

Also using a 25mm GPS Antenna along with those real nice cases you desgined on Thingiverse. Great work on those cases!

garthvh, 2022-08-14:
Once I used the power button to power off and back on instead of the reset button I got GPS back

tropho23, 2022-08-14:
> > Same for me; after installing 1.3.37 I also see a 'No GPS' message on my Tbeam 1.1 with NEO-6M GPS chipset. I'm using a Cirocomm 25mm GPS antenna but that ain't helping me here.
> 
> Also using a 25mm GPS Antenna along with those real nice cases you desgined on Thingiverse. Great work on those cases!

Thanks!

caveman99, 2022-08-16:
the device will attempt to factory reset the GPS Chip if it doesn't detect valid NMEA sentences after 60 seconds. That can happen on a direct change from 1.2 to 1.3. The Reset will require a device reboot though. The state is saved in a proto file, that process may have been interrupted during the latest littlefs debugging attempts.

1n5aN1aC, 2022-08-19:
I just tested `firmware-tbeam-1.3.38.1253abd` on a device marked `T22_V1.1` and after setting Region & Owner via the Python-CLI, Also am getting a "No GPS Module" message on both devices.

I did try letting them set 5 minutes or so, then rebooting via both the power button, via the reset button, or the python-cli, but there was no change.

I will also note that both shortly had the little tiny red led blinking that usually indicates a GPS fix by the module....

ndoolin, 2022-08-19:
> I just tested `firmware-tbeam-1.3.38.1253abd` on a device marked `T22_V1.1` and after setting Region & Owner via the Python-CLI, Also am getting a "No GPS Module" message on both devices.
>
> I did try [...]

I eventually got one of my Tbeams working.  I ended up watching this video and downloaded the GPS reset code referenced in the video and uploaded to my Tbeams. https://youtu.be/xVaWinSfsz4  It worked for one, but not the other.  My GPS red light would also blink suggesting it has a GPS lock.   Additionally I was using one of the Tbeams to map the helium network and Re flashing Helium Mapper code brings the GPS back to life, however, if I reflash Meshtastic it once again goes back to  No GPS.  I don’t think anything is wrong with the hardware but as others have suggested it’s more of a GPS sentence format that seems to somehow be getting jumbled up.

jaycap97, 2022-09-11:
Does anyone know what the last version does the GPS work for TBeams?

garthvh, 2022-09-11:
You need to power cycle your device twice with the power button, wait 5 minutes between each.

pcopa, 2022-09-13:
Me Too,

I have two modules that I've regularly reflashed to the latest version and these two are working fine.
A few days ago, I received my third module that showed "NO GPS" out of the box. I then reflashed it to the same version as my other 2 (1.3.39) but "no joy".  
Power cycling the device twice with the power button, wait 5 minutes between each, did not make a difference for me.

code8buster, 2022-09-15:
Whatever tricks worked for all of you, nothing worked for my NEO6M Tbeam.
The last pull request I flashed which worked was artifact pr1593-firmware-1.3.29.2ad9e23
I had also tried firmware 1.3.39 which did not work, exhibiting the same non functioning described in the issue.

pcopa, 2022-09-16:
I did get it to work once after flashing the 'GPS reset' from'[https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware/GPS_%20reset_test?spm=a2g0s.imconversation.0.0.71673e5fxJiloO](url) and then flashing 'firmware-tbeam1.1-1.2.60.ab959de.bin'.  The good news is that I got a third screen that went from 'NO GPS' to 'NO GPS LOCK' even though the RED LED is always blinking after ~15 sec in all versions.
But today I restated my 1.3.40 version and I'm back to 'NO GPS' and only two screens.

ndoolin, 2022-09-16:
Same think I’ve seen on 1.39.  I’ve had it eventually show “No GPS Lock” at points as well, but it never actually indicates a lock, even though it appears to have a lock based on the LED behavior.-Nick

asfarley, 2022-09-18:
Following up on this: 

I have 3 T-Beam v1.1, just recently arrived from Shenzhen. All three initially had the "No GPS Module" symptom. I flashed all three with the GPS Reset V3 sketch, and it seemed to do the trick; the NEO-6M units all started printing out NMEA data at this point. 

I re-flashed them with Meshtastic firmware 1.3.40, Now they all say "No GPS Lock", which (I suppose) indicates that the unit thinks it's getting valid NMEA data, but the data is telling it that it can't see satellites. Strangely, the red GPS LED is blinking at 1Hz on all three units suggesting there is a lock. 

I went back to the GPS-reset V3 sketch and connected with TerraTerm. I can see that the NEO-6M actually does have a lock, even in my relatively terrible basement window! 

This suggests to me that the 1.3.40 firmware is not parsing the NMEA stream correctly or something like that. It might be nice if there was a mode to ask the Meshtastic firmware itself what it thinks it's seeing from the Ublox, because it's unclear to me why it isn't able to parse the stream. 

Here is what I got from the GPS reset sketch firmware:

```
$GPGGA,211022.00,5104.68369,N,11403.80435,W,1,08,1.23,1072.3,M,-17.5,M,,*52
$GPGGA,211023.00,5104.68359,N,11403.80432,W,1,08,1.23,1072.1,M,-17.5,M,,*55
$GPGGA,211024.00,5104.68362,N,11403.80460,W,1,08,1.23,1072.2,M,-17.5,M,,*5E
$GPGGA,211025.00,5104.68336,N,11403.80545,W,1,08,1.23,1071.4,M,-17.5,M,,*5D
```

Hello from Calgary, Alberta.

garthvh, 2022-09-19:
Fixed in:

https://github.com/meshtastic/Meshtastic-device/releases/tag/v1.3.41.80ddb81

closed 2022-09-19 by garthvh

1n5aN1aC, 2022-09-19:
I can confirm this appears to have fixed the problem on 5 of my TBeam modules.  (1.3.41.80ddb81)

asfarley, 2022-09-19:
Is everyone able to change settings e.g. Lora region after this update? The iphone app crashes when I click on the Lora sub-menu for me after this. The app appears to still be 1.3.40.

1n5aN1aC, 2022-09-19:
> Is everyone able to change settings e.g. Lora region after this update? The iphone app crashes when I click on the Lora sub-menu for me after this. The app appears to still be 1.3.40.

I used the python client via USB to program all mine.

I noted that this update, after flashing, I seem to have to unplug USB, hard power off the module, power back on, plug USB, and then I can set region, name, etc.

I would assume you could set these via the app as well after a reboot:  flash, hard power off module, power on module, connect, configure.     But note that I have not tested this.

asfarley, 2022-09-19:
@1n5aN1aC Thanks, I tried that - the Python client is giving me "Timed out waiting for connection completion". I have tried the full reboot cycle.

ndoolin, 2022-09-19:
Had the same issue.pip install —upgrade Meshtastic —prethen…  meshtastic —set lora.region <your region>Still “No GPS” for me on 1.3.41

asfarley, 2022-09-19:
Updating the cli as suggested by @ndoolin worked for me and now my devices all appear to be functioning well via the CLI. Looking forward to trying the new iPhone app!

pcopa, 2022-09-24:
all mine are working with 1.3.41
Thanks

Kaeteker, 2022-09-24:
I updated my tbeams from 1.2.x (can't remember exact version) to 1.3.41 and I think I'm experiencing this issue. I have two tbeam v1.0, and both of them report "no GPS module" after the update. The amber GPS led never blinks. I tried using the GPS-reset-v3 sketch, and that was able to get a lock and output my coordinates to the serial monitor. When I then reflashed with 1.3.41, it reported "no GPS lock" instead of "no GPS module", managed to get a lock, and worked perfectly normally until I unpowered the device and turned it back on. Then it went back to "no GPS module." The GPS stays broken until I reflash with the reset sketch, and immediately breaks again as soon as the device loses power. I don't have a tbeam v1.1 to see if this is specific to something I'm doing wrong, or to the tbeam v1.0.

garthvh, 2022-09-25:
Have you power cycled (not reset)twice waiting 5 minutes in between? It has fixed every device I have tried it on.

Kaeteker, 2022-09-25:
By power cycling, do you mean turning it off by holding the power button, then back on, like was said in a comment further up? If so, yeah I've done that. Turned it on, waited >5 minutes, turned it off then back on, waited >5 minutes, repeated that a few times on each device in hopes it would work. Hasn't made a difference, only way I've found to get GPS working is by using the reset-v3 sketch and then making sure the device doesn't lose power while I want to use it.

pcopa, 2022-09-25:
@Kaeteker : what I usually do after I've re-flashed and it doesn't work, I will:

- Hold the middle button down until the device shuts off,
- Remove the USB cable and remove the battery, just to be Shure all power is off.
- Wait >5 minutes to ensure all caps are discharged.
- Then insert the (powered) USB cable and the device should startup on its own.

Just my observations:

1.  If after bootup, the bottom line of the display only has two dots, then the device will be stuck on "NO GPS" and no commands will get it out of that mode. Only re-flashing got it work for me.
2. If after bootup, the bottom line of the display has three dots, then the device will show "NO GPS LOCK" and this means there is life.  After a re-flash and get my first lock, I had to go outside and wait ~5 minutes. After that I can go back into the basement and get a lock every time.

Good luck

ajorg, 2022-11-07:
I tried the power cycle thing on my TBeam on 2.0.0 to no avail. The [`GPS_ reset.bin`](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/firmware/GPS_%20reset_test/GPS_%20reset.bin) firmware from LilyGo did get it going again, though.

So it seems, to me at least, that the attempts to reset the GPS from the Meshtastic firmware aren't working. Wondering if this should be re-opened, but my own problem is solved, so I won't push for it.

SongHoang, 2022-11-09:
Same, still having issues on the latest firmware. Have to use TTGO reset .ino, then it'll work. But weirdly, after turning on and off, GPS will stop working and show No GPS Module.

ufanders, 2023-06-16:
My T-Beam v1.1 boards show the same symptom - "No GPS Lock" with two dots on the display. Was this issue ever fully resolved?

garthvh, 2023-06-16:
No GPS Lock is not the same error, your device GPS had been detected and is working and has not managed to get a lock.

Maxwell190420, 2023-07-01:
What I did is: 
Go to Radio Configuration
    Change some parameter, e.g. Position broadcast interval and send the new setup.
    And it works. The rest is fine. Even Rx Tx GPS pins
The problem I've found is App do not send config setup. You have to do it.
Don't know if it work in other cases.