meshtastic/firmware #1637: [Bug]:

arnotron, 2022-08-18:
### Category

Other

### Hardware

T-Echo

### Firmware Version

1.3.38.1253abd

### Description

It would be helpful to have the current date+time displayed in the status screen of at least my T-Echo devices. 

The reason is that I am observing strange position history information of +5 years age in the iOS app, the log states 913120360 which translates to Tue Dec  8 13:32:40 CET 1998 and I am not sure which part is using the wrong time stamp. Also, I think it does not hurt to display this information.

### Relevant log output

```Shell
12:32:40 32 [EnvironmentTelemetryModule] -----------------------------------------
12:32:40 32 [EnvironmentTelemetryModule] Environment Telemetry: Read data
12:32:40 32 [EnvironmentTelemetryModule] BME280Sensor::getMetrics
12:32:40 32 [EnvironmentTelemetryModule] Telemetry->time: 913120360
12:32:40 33 [EnvironmentTelemetryModule] Telemetry->barometric_pressure: 1013.671692
12:32:40 33 [EnvironmentTelemetryModule] Telemetry->current: 0.000000
12:32:40 33 [EnvironmentTelemetryModule] Telemetry->gas_resistance: 0.000000
12:32:40 33 [EnvironmentTelemetryModule] Telemetry->relative_humidity: 40.062500
12:32:40 33 [EnvironmentTelemetryModule] Telemetry->temperature: 38.540001
12:32:40 33 [EnvironmentTelemetryModule] Telemetry->voltage: 0.000000
12:32:40 33 [EnvironmentTelemetryModule] Environment Telemetry: Sending packet to meshUpdate DB node 0x6fce0454, rx_time=0
12:32:40 33 [EnvironmentTelemetryModule] handleReceived(LOCAL) (id=0x45555d36 Fr0x54 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=67)
12:32:40 33 [EnvironmentTelemetryModule] No modules interested in portnum=67, src=LOCAL
12:32:40 33 [EnvironmentTelemetryModule] Add packet record (id=0x45555d36 Fr0x54 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=67)
12:32:40 33 [EnvironmentTelemetryModule] Expanding short PSK #1
12:32:40 33 [EnvironmentTelemetryModule] Using AES128 key!
12:32:40 33 [EnvironmentTelemetryModule] nRF52 encrypt fr=6fce0454, num=45555d36, numBytes=26!
12:32:40 33 [EnvironmentTelemetryModule] enqueuing for send (id=0x45555d36 Fr0x54 To0xff, WantAck0, HopLim3 Ch0x6e encrypted)
12:32:40 33 [EnvironmentTelemetryModule] txGood=0,rxGood=0,rxBad=0
12:32:40 33 [EnvironmentTelemetryModule] Using channel 0 (hash 0x6e)
12:32:40 33 [EnvironmentTelemetryModule] Expanding short PSK #1
12:32:40 33 [EnvironmentTelemetryModule] Using AES128 key!
12:32:40 33 [EnvironmentTelemetryModule] nRF52 encrypt fr=6fce0454, num=45555d36, numBytes=26!
12:32:41 33 [EnvironmentTelemetryModule] decoded message (id=0x45555d36 Fr0x54 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=67 priority=64)
```

garthvh, 2022-08-18:
GPS of the techo is the source, iOS displays time from the device without modification, or unknown age for dates more than six months ago, the device screen does the same.

andrekir, 2022-08-18:
seeing the same with T-Echo on 1.3.38. GPS with 5+ sat lock and good location alt/lat/lon.

@arnotron there is GMT time (but no date) in the last tab (2nd line under batt %). minutes off are a sure indication of bad data.

garthvh, 2022-09-11:
@lewisxhe any idea why we get 5 year old dates from the techo gps?

lewisxhe, 2022-09-11:
@arnotron Is the satellite already locked? I can't see anything else about GPS from the logs

arnotron, 2022-09-11:
Sorry - GitHub GUI confused me. 
The T-Echo was locked to the gps satellite, otherwise it wouldn’t be able to provide these locations. I will update tongue ltest version and try to get a more detailed bug report ready.

garthvh, 2022-09-11:
I did update the ui in the iOS app to show unknown time for the pins like the other places time is shown when it is > 6 months old, which matches the device code.  Seems to me that the t-echo gps uses 5 years ago as null instead of unix epoch like our other GPS modules.

caveman99, 2022-09-12:
That could also be interaction or bad data from the T-Echo RTC Chip. If there's a known RTC chip on the device we use it and only update it from GPS every 24 hours. Currently impemented for T-Echo and the RAK RTC

caveman99, 2022-10-04:
addressed in #1729

closed 2022-10-04 by caveman99