meshtastic/firmware #1643: [Bug]: Web content missing on 1.3.39.ddc3727

rohanki, 2022-08-24:
### Category

Other

### Hardware

T-Lora v1.3

### Firmware Version

1.3.39.ddc3727

### Description

T-Lora v1.3 running 1.3.39.ddc3727 on WiFi - meshtastic-web is missing from this build..
Accessing the device over wifi returns:

Web server is running.

The content you are looking for can't be found. Please see: [FAQ](https://meshtastic.org/docs/getting-started/faq#wifi--web-browser).

[admin]

going to admin/manage web content returns a 404
404 Not Found

### Relevant log output

```Shell
noproto output
[E][vfs_api.cpp:64] open(): /littlefs/static/index.html does not exist
[E][vfs_api.cpp:64] open(): /littlefs/static/index.html.gz does not exist
[E][vfs_api.cpp:64] open(): /littlefs/static/index.html.gz does not exist
```

t413, 2022-09-21:
I'm getting this as well. 

Apparently it's either BLE or WiFi .. and you can't switch back to BLE operation without the web interface which is .. gone.

caveman99, 2022-09-27:
This will be fixed in 1.3.42

closed 2022-09-27 by caveman99