meshtastic/firmware #1644: [Bug]: panic

sircambridge, 2022-08-24:
### Category

Other

### Hardware

T-Beam

### Firmware Version

1.3.39

### Description

Code was running fine with 1.3.33 on T-beam v1.1 meshtastic.

I updated to 1.3.39 today and immediately go into a panic loop. going back to 1.3.33 fixes the problem!

sorry I wish I had more details but i dont quite know how to debug this. all I know is that output from serial output.

### Relevant log output

```Shell
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060530  A0      : 0x800f5b8b  A1      : 0x3ffd5730  
A2      : 0x3ffd5f00  A3      : 0x00000000  A4      : 0x3ffc3124  A5      : 0x00008f80  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800f587a  A9      : 0x3ffd5720  
A10     : 0x3ffd5ea4  A11     : 0x3ffd5f40  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3ff47024  A15     : 0x00000040  SAR     : 0x0000000a  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x00000000:0x3ffd5730 0x400f5b88:0x3ffd5750 0x400f5c32:0x3ffd5770 0x400d9cdf:0x3ffd57a0 0x400da0ee:0x3ffd57c0 0x400d5682:0x3ffd57f0 0x400f6cad:0x3ffd5810 0x400dd658:0x3ffd5840 0x4012bdad:0x3ffd5860

Rebooting...
```

thebentern, 2022-08-24:
Was the device flashed via device-install.bat or device-install.sh script? This release has a new partitioning scheme for the esp32 boards.

sircambridge, 2022-08-24:
No, I just did a git pull from master from github, and ran compile and upload in platform I/O.

I will look into using device-install.bat and report back thanks.

sircambridge, 2022-08-24:
Em not super familiar with this codebase. is device-install.bat part of this repo? I'm having trouble figuring this out.  can i get a pointer in the right direction as to where to find it and run it?

garthvh, 2022-08-24:
> I'm not super familiar with this codebase. is device-install.bat part of this repo? I'm having trouble figuring this out. can i get a pointer in the right direction as to where to find it and run it?

They are included in the firmware release download zip file.

thebentern, 2022-08-24:
@caveman99, would you mind taking a look at our platformio setu in lifht of the partition change? I'm having issues with esp32 devices after it (https://github.com/meshtastic/Meshtastic-device/commit/d6d936b5d2e5f8960092a0055fd64c1fc79dc63b) as well with just a a standard upload. My updated device-install scripts wipe those sections of flash explicitly however and seem to work fine.

sircambridge, 2022-08-27:
I tried reverting this commit (https://github.com/meshtastic/Meshtastic-device/commit/d6d936b5d2e5f8960092a0055fd64c1fc79dc63b) on my local partitions-table.csv but i still get the panic

caveman99, 2022-09-24:
can you please double check with the latest build? https://github.com/meshtastic/Meshtastic-device/suites/8445496242/artifacts/374421783

thebentern, 2022-09-25:
Closing this. On the latest releases cant repro after fresh flash on multiple permutation of T-Beams. 
Please comment if you have issues with the current release.

closed 2022-09-25 by thebentern