meshtastic/firmware #1646: Bug: Long hold of middle button will crash tbeam if there's no screen.

mc-hamster, 2022-08-24:
Bug: Long hold of middle button will crash tbeam if there's no screen. This is on 1.3.39.

100% reproducible.

garthvh, 2022-08-25:
Mine goes to AP mode, that is the long press action right?

//\ E S H T /\ S T / C

??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=reset
??:??:?? 0 Setting forceSoftAP = 1
??:??:?? 0 Filesystem files:
??:??:?? 0   /prefs/channels.proto (53 Bytes)
??:??:?? 0   /prefs/config.proto (40 Bytes)
??:??:?? 0   /prefs/db.proto (145 Bytes)
??:??:?? 0   /prefs/module.proto (16 Bytes)
??:??:?? 0 I2C device found at address 0x34
??:??:?? 0 axp192 PMU found
??:??:?? 0 1 I2C devices found
??:??:?? 0 Meshtastic hwvendor=4, swver=1.3.39.ddc3727-d
??:??:?? 0 Setting random seed 3180027834
??:??:?? 0 Total heap: 239484
??:??:?? 0 Free heap: 207848
??:??:?? 0 Total PSRAM: 4194252
??:??:?? 0 Free PSRAM: 4194252
??:??:?? 0 NVS: UsedEntries 117, FreeEntries 513, AllEntries 630, NameSpaces 4
??:??:?? 0 Setup Preferences in Flash Storage
??:??:?? 0 Number of Device Reboots: 6
??:??:?? 0 Initializing NodeDB
??:??:?? 0 Loading /prefs/db.proto
??:??:?? 0 Loaded saved devicestate version 15
??:??:?? 0 Loading /prefs/config.proto
??:??:?? 0 Loaded saved config version 15
??:??:?? 0 Loading /prefs/module.proto
??:??:?? 0 Loaded saved moduleConfig version 15
??:??:?? 0 Loading /prefs/channels.proto
??:??:?? 0 Loaded saved channelFile version 15
??:??:?? 0 Number of Device Reboots: 6
??:??:?? 0 Expanding short PSK #1
??:??:?? 0 Wanted region 0, using Unset
??:??:?? 0 region=0, NODENUM=0xf71e89f0, dbsize=1
??:??:?? 0 Saving /prefs/db.proto
??:??:?? 0 Saving /prefs/config.proto
??:??:?? 0 Saving /prefs/module.proto
??:??:?? 0 Saving /prefs/channels.proto
chip id detect 0x3
Detect CHIP :AXP192
OUTPUT Register 0x5f
??:??:?? 0 AXP192 Begin PASS
??:??:?? 0 DCDC1: ENABLE
??:??:?? 0 DCDC2: ENABLE
??:??:?? 0 LDO2: ENABLE
??:??:?? 0 LDO3: ENABLE
??:??:?? 0 DCDC3: ENABLE
??:??:?? 0 Exten: ENABLE
??:??:?? 0 ----------------------------------------
??:??:?? 1 DCDC1: ENABLE
??:??:?? 1 DCDC2: ENABLE
??:??:?? 1 LDO2: ENABLE
??:??:?? 1 LDO3: ENABLE
??:??:?? 1 DCDC3: ENABLE
??:??:?? 1 Exten: ENABLE
SRC REG:0xc4
Charging enable is enable
Charging target-voltage : 0x2
 end when the charge current is lower than 10% of the set value
Charge current : 700.00 mA
??:??:?? 1 Battery: usbPower=1, isCharging=1, batMv=3909, batPct=68
??:??:?? 1 Read RTC time as 1661436273
??:??:?? 1 Using MSL altitude model
??:??:?? 1 WANT GPS=1
??:??:?? 1 GxGSA NOT available
??:??:?? 1 External Notification Module Disabled
??:??:?? 1 Starting meshradio init...
??:??:?? 1 Set radio: region=Unset, name=LongF, config=0, ch=91, power=30
??:??:?? 1 Radio myRegion->freqStart / myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
??:??:?? 1 Radio myRegion->numChannels: 104
??:??:?? 1 Radio channel_num: 91
??:??:?? 1 Radio frequency: 924.875000
??:??:?? 1 Slot time: 42 msec
??:??:?? 1 Set radio: final power level=20
??:??:?? 1 Current limit set to 100.000000
??:??:?? 1 Current limit set result 0
??:??:?? 1 RF95 init result -19
??:??:?? 1 Frequency set to 924.875000
??:??:?? 1 Bandwidth set to 250.000000
??:??:?? 1 Power output set to 20
??:??:?? 1 Current limit set to 100.000000
??:??:?? 1 Current limit set result 0
??:??:?? 1 RF95 Radio init succeeded, using RF95 radio
??:??:?? 1 WiFi ... Forced AP Mode
??:??:?? 1 Waiting [E][Preferences.cpp:457] getBytesLength(): nvs_get_blob len fail: PK NOT_FOUND
[E][Preferences.cpp:457] getBytesLength(): nvs_get_blob len fail: cert NOT_FOUND
for SSL Cert to be generated.
??:??:?? 25 SSL Cert Ready!
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
??:??:?? 26 ************ [WiFi-event] event: 0 ************
??:??:?? 26 WiFi interface ready[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START

??:??:?? 26 Starting (Forced) WIFI AP: ssid=meshtasticAdmin, ok=1
??:??:?? 26 mDNS responder started
??:??:?? 26 mDNS Host: Meshtastic.local
??:??:?? 26 MY IP AP ADDRESS: 192.168.4.1
??:??:?? 26 Starting NTP time client
??:??:?? 26 Initializing Web Server ...
??:??:?? 26 Starting Secure Web Server...
??:??:?? 26 Starting Insecure Web Server...
??:??:?? 26 Web Servers Ready! :-) 
??:??:?? 26 API server listening on TCP port 4403
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
??:??:?? 26 ************ [WiFi-event] event: 14 ************
??:??:?? 26 WiFi access point started
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
 26 ************ [WiFi-event] event: 14 ************
??:??:?? 26 WiFi access point started

caveman99, 2022-08-25:
Maybe the crash is trying to enable wifi while paired with bluetooth? I can't reproduce with a standalone device either

garthvh, 2022-08-25:
> Maybe the crash is trying to enable wifi while paired with bluetooth? I can't reproduce with a standalone device either

I wonder if his AXP chip is fried

mc-hamster, 2022-08-25:
There's an order of operation. Middle button will only activate wifi if it's held down within the first (about) quarter second of boot. Middle button won't activate wifi otherwise.

garthvh, 2022-08-25:
There is a crash if you wait before the long press 

meshtastic --noproto
Warning: Multiple serial ports were detected so one serial port must be specified with the '--port'.
  Ports detected:['/dev/cu.usbmodem11301', '/dev/cu.usbmodem53230057621', '/dev/cu.wchusbserial53230057621']
garthvanderhouwen@Garths-Mac-mini ~ % meshtastic --noproto
WARNING file:mesh_interface.py _sendToRadio line:467 Not sending packet because protocol use is disabled by noProto
Connected to radio
WARNING file:mesh_interface.py _sendPacket line:357 Not sending packet because protocol use is disabled by noProto
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x401f2af0  PS      : 0x00060930  A0      : 0x800f4c96  A1      : 0x3ffd5670  
A2      : 0x3ffd59f8  A3      : 0x00000180  A4      : 0x00000002  A5      : 0x00000004  
A6      : 0x3ffc3000  A7      : 0x3ffc2f8c  A8      : 0x000000c0  A9      : 0x00000001  
A10     : 0xffffffff  A11     : 0x3ffc3008  A12     : 0x400dd524  A13     : 0x3ffd5690  
A14     : 0x00000000  A15     : 0x3ffd3860  SAR     : 0x00000002  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000180  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x401f2af0:0x3ffd5670 0x400f4c93:0x3ffd5690 0x400dc433:0x3ffd56b0 0x400dd645:0x3ffd56e0 0x400f6086:0x3ffd5700 0x400f60f6:0x3ffd5720 0x400dd535:0x3ffd5740 0x400d56ae:0x3ffd5760 0x400f6f31:0x3ffd5780 0x400dd88c:0x3ffd57b0 0x4012c01d:0x3ffd57d0

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
WARNING file:mesh_interface.py _sendToRadio line:467 Not sending packet because protocol use is disabled by noProto
??:??:?? 0 

//\ E S H T /\ S T / C

??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=reset
??:??:?? 0 Setting forceSoftAP = 1
??:??:?? 0 Filesystem files:
??:??:?? 0   /prefs/channels.proto (53 Bytes)
??:??:?? 0   /prefs/config.proto (42 Bytes)
??:??:?? 0   /prefs/db.proto (330 Bytes)
??:??:?? 0   /prefs/module.proto (16 Bytes)
??:??:?? 0 I2C device found at address 0x34
??:??:?? 0 axp192 PMU found
??:??:?? 0 1 I2C devices found
??:??:?? 0 Meshtastic hwvendor=4, swver=1.3.40.3988f14-d
??:??:?? 0 Setting random seed 3701954928
??:??:?? 0 Total heap: 239472
??:??:?? 0 Free heap: 207808
??:??:?? 0 Total PSRAM: 4194252
??:??:?? 0 Free PSRAM: 4194252
??:??:?? 0 NVS: UsedEntries 102, FreeEntries 528, AllEntries 630, NameSpaces 5
??:??:?? 0 Setup Preferences in Flash Storage
??:??:?? 0 Number of Device Reboots: 10
??:??:?? 0 Initializing NodeDB
??:??:?? 0 Loading /prefs/db.proto
??:??:?? 0 Loaded saved devicestate version 15
??:??:?? 0 Loading /prefs/config.proto
??:??:?? 0 Loaded saved config version 15
??:??:?? 0 Loading /prefs/module.proto
??:??:?? 0 Loaded saved moduleConfig version 15
??:??:?? 0 Loading /prefs/channels.proto
??:??:?? 0 Loaded saved channelFile version 15
??:??:?? 0 Number of Device Reboots: 10
??:??:?? 0 Expanding short PSK #1
[... 155 lines cut ...]
21:07:52 40 [Router] old user !f71e89f0/Meshtastic 89f0/89f0
21:07:52 40 [Router] updating changed=0 user !f71e89f0/Meshtastic 89f0/89f0
21:07:52 40 [Router] Module 'nodeinfo' considered
21:07:52 40 [Router] Module 'routing' wantsPacket=1
21:07:52 40 [Router] Received routing from=0xf71e89f0, id=0x70e88953, portnum=4, payloadlen=44
21:07:52 40 [Router] Routing sniffing (id=0x70e88953 Fr0xf0 To0x88, WantAck0, HopLim3 Ch0x0 Portnum=4 requestId=2dc174cb rxtime=1661461672 rxSNR=9 rxRSSI=1.73226)
21:07:52 40 [Router] Received an ack for 0x2dc174cb, stopping retransmissions
21:07:52 40 [Router] FIXME-update-db Sniffing packet
21:07:52 40 [Router] Delivering rx packet (id=0x70e88953 Fr0xf0 To0x88, WantAck0, HopLim3 Ch0x0 Portnum=4 requestId=2dc174cb rxtime=1661461672 rxSNR=9 rxRSSI=1.73226)
21:07:52 40 [Router] Forwarding to phone (id=0x70e88953 Fr0xf0 To0x88, WantAck0, HopLim3 Ch0x0 Portnum=4 requestId=2dc174cb rxtime=1661461672 rxSNR=9 rxRSSI=1.73226)
21:07:52 40 [Router] Update DB node 0xf71e89f0, rx_time=1661461672
21:07:52 40 [Router] Module 'routing' considered
21:07:54 42 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=66, time 1148 ms
21:07:54 42 [RadioIf] ignoring received packet due to error=-7
21:07:54 42 [RadioIf] AirTime - Packet received (noise?) : 1148ms
21:07:56 43 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=44, time 886 ms
21:07:56 43 [RadioIf] Lora RX (id=0x2dc174cc Fr0x88 To0xff, WantAck0, HopLim2 Ch0x6e encrypted rxSNR=10.75 rxRSSI=1.73226)
21:07:56 43 [RadioIf] AirTime - Packet received : 886ms
21:07:56 43 [Router] Rx someone rebroadcasting for us (id=0x2dc174cc Fr0x88 To0xff, WantAck0, HopLim2 Ch0x6e encrypted rxSNR=10.75 rxRSSI=1.73226)
21:07:56 43 [Router] didn't find pending packet
21:07:56 43 [Router] Found existing packet record for fr=0x9387ac88,to=0xffffffff,id=0x2dc174cc
21:07:56 43 [Router] Found existing packet record for fr=0x9387ac88,to=0xffffffff,id=0x2dc174cc
21:07:56 43 [Router] Add packet record (id=0x2dc174cc Fr0x88 To0xff, WantAck0, HopLim2 Ch0x6e encrypted rxSNR=10.75 rxRSSI=1.73226)
21:07:56 43 [Router] Ignoring incoming msg, because we've already seen it (id=0x2dc174cc Fr0x88 To0xff, WantAck0, HopLim2 Ch0x6e encrypted rxSNR=10.75 rxRSSI=1.73226)
21:07:56 43 [Router] Incoming message was filtered 0x9387ac88

garthvh, 2022-08-25:
After the initial crash if you are still holding the button it goes into AP mode

commit fefcbb14 (2022-10-04): fix #1646
closed 2022-10-04 by caveman99