meshtastic/firmware #1647: Bug: Device power-down doesn't work on unset screen

mc-hamster, 2022-08-24:
When the unset screen is displayed, holding down the middle button on tbeam will not power down the node.

This is on 1.3.39.

thebentern, 2022-08-25:
Hmmm, holding down the PWR button for 10 seconds is working on my node with the region unset screen.

garthvh, 2022-08-25:
Same, I have two devices where I can pull the screen and both are behaving properly, is your AXP dead on the board in question?

garthvh, 2022-08-25:
I updated the title, my device shuts down but the text of the last screen displayed stays on the screen

garthvh, 2022-09-28:
04:22:38 730 [Button] Long press start!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x401f5e28  PS      : 0x00060530  A0      : 0x800f8386  A1      : 0x3ffd56f0  
A2      : 0x3ffd59b8  A3      : 0x00000180  A4      : 0x00000002  A5      : 0x00000004  
A6      : 0x3ffc2fe4  A7      : 0x3ffc2f70  A8      : 0x000000c0  A9      : 0x00000001  
A10     : 0xffffffff  A11     : 0x3ffc2fec  A12     : 0x400e01e0  A13     : 0x00000005  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000002  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000180  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x401f5e28:0x3ffd56f0 0x400f8383:0x3ffd5710 0x400df0d7:0x3ffd5730 0x400e02fd:0x3ffd5760 0x400f9776:0x3ffd5780 0x400f97e6:0x3ffd57a0 0x400e01f1:0x3ffd57c0 0x400d82ea:0x3ffd57e0 0x400fa621:0x3ffd5800 0x400e0544:0x3ffd5830 0x4012f169:0x3ffd5850

thebentern, 2022-09-28:
Output:

 ```
   0x400e01e0: recordCriticalError(_CriticalErrorCode, unsigned int, char const*) at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mesh/NodeDB.cpp:669
    0x400f8383: pb_write at /home/runner/work/Meshtastic-device/Meshtastic-device/.pio/libdeps/tbeam/Nanopb/pb_encode.c:87
    0x400df0d7: MeshModule::allocAckNak(_Routing_Error, unsigned int, unsigned int, unsigned char) at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mesh/MeshModule.cpp:40
    0x400e02fd: NodeDB::updatePosition(unsigned int, _Position const&, RxSource) at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mesh/NodeDB.cpp:517
    0x400f9776: SX126x::standby() at /home/runner/work/Meshtastic-device/Meshtastic-device/.pio/libdeps/tbeam/RadioLib/src/modules/SX126x/SX126x.cpp:364
    0x400f97e6: SX126x::setRx(unsigned int) at /home/runner/work/Meshtastic-device/Meshtastic-device/.pio/libdeps/tbeam/RadioLib/src/modules/SX126x/SX126x.cpp:1314
    0x400e01f1: recordCriticalError(_CriticalErrorCode, unsigned int, char const*) at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mesh/NodeDB.cpp:675
    0x400d82ea: NMEAGPS::lookForLocation() at /home/runner/work/Meshtastic-device/Meshtastic-device/src/gps/NMEAGPS.cpp:179
    0x400fa621: SX126x::startTransmit(unsigned char*, unsigned int, unsigned char) at /home/runner/work/Meshtastic-device/Meshtastic-device/.pio/libdeps/tbeam/RadioLib/src/modules/SX126x/SX126x.cpp:387
    0x400e0544: loadProto(char const*, unsigned int, unsigned int, pb_msgdesc_s const*, void*) at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mesh/NodeDB.cpp:304
    0x4012f169: __hexnan at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/gdtoa-hexnan.c:99

```

dmg244, 2022-10-18:
Confirm,  2pcs T-Beam with M8N & SX1262 with the 1.3" OLED shuts down but the text of the last screen displayed stays on the screen. FW v1.3.42.9bd9252.
 If install FW v1.3.43.aae9d2f  or v1.3.45.b0d0552 I have garbage on the screen.

garthvh, 2023-01-24:
Not seeing this anymore if the OLED screen type is set

closed 2023-01-24 by garthvh