meshtastic/firmware #1661: 1.3: MQTT is broken

Noki, 2022-08-29:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

1.3.40.FD27A40-D

### Description

The MQTT functionality is broken in 1.3. In the current state only the online/offline status for the node is working.

When reworking the MQTT functionality https://github.com/meshtastic/Meshtastic-device/issues/1100 should be addressed as well.

garthvh, 2022-08-30:
Are you using your own MQTT server or the public one?

Noki, 2022-08-30:
The public one.

garthvh, 2022-08-31:
After setting up wifi and enabling MQTT

meshtastic --set mqtt.enabled true

And enabling downlink and uplink

meshtastic --ch-set downlink_enabled true --ch-set uplink_enabled true --ch-index 0

I am seeing JSON MQTT payloads in the serial log.

closed 2022-08-31 by garthvh

Noki, 2022-08-31:
Just as a note: The enabled/disabled status indicator of the Web Client is currently indicating the opposite, as the option was renamed from mqtt.disabled to mqtt.enabled. This got me confused.

caveman99, 2022-08-31:
@sachaw just fixed that in meshtastic-web, thanks for the pointer.