meshtastic/firmware #1665: Feature Request: Imperial units option for ESP32 display

ghostop14, 2022-08-31:
Would be nice to have a toggle in the display configuration to change the ESP32 screen display units from metric to imperial.

sachaw, 2022-09-11:
Protobufs implemented

ajmcquilkin, 2022-09-21:
@ghostop14 New to this repo but potentially interested in taking this on. Do you have a list of displayed measurements that would need to be converted?

ghostop14, 2022-09-21:
This may actually be 3 issues.  
1. On the device LCD screen, currently it shows km (that was the one I was originally thinking of, as a device setting)
2. In the iOS app, if you go under nodes, distance is in ft (imperial).  If we had a global imperial/metric, this should probably toggle with that
3. On the android app in the people (node) section, distance is not displayed at all

ajmcquilkin, 2022-09-21:
Yeah that definitely seems like three issues. I'm happy to take the new `DisplayConfig.DisplayUnits` config field and convert metric units into imperial for display on the screen based on that value, assuming that a user would be able to set that field within the app or the cli

mc-hamster, 2022-09-22:
right on! freedom units!

closed 2022-09-22 by sachaw

garthvh, 2022-09-24:
iOS is based on your phone settings, not the setting for the device screen.

PR #1717 (2022-09-22): Add Support for Imperial Units on Screen
This PR adds support for imperial units within on-device screens based on the `config.display.units` field. Metric units are the default, and this PR allows users to override this default. This change currently applies to the following fields:

- Altitude
- Distance to target

Closes  #1665