meshtastic/firmware #1666: [Bug]: Screen on Nano G1 flips out on initial flash

CowboyPilot, 2022-09-01:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

1.3.39ddc3727

### Description

After initial boot on Nano G1 the screen is garbled.  Mostly snow except for about 3 horizontal lines.  Changing the region seems to restore the screen functionality but after a subsequent boot the screen comes back as garbled.
[image]

garthvh, 2022-09-02:
A pull from @neilhao is incoming for this issue

neilhao, 2022-09-02:
This problem is expected to be fixed with #1670

sachaw, 2022-09-03:
This is now merged, please reopen if this is now fixed

closed 2022-09-03 by sachaw

CowboyPilot, 2022-09-03:
Just tested it, works great!