meshtastic/firmware #1675: [Bug]: positionBroadcastSecs to non 0 value causes erratic

rohanki, 2022-09-06:
### Category

Other

### Hardware

T-Beam, T-Lora v1.3, T-Lora v2 1.6

### Firmware Version

1.3.39.ddc3727

### Description

Setting positionBroadcastSecs to a value other than 0 causes PositionModule to send position updates continuously and erratically (approx 5-10 seconds in between updates) regardless of value set. (I also tried setting the value using ms just to make sure this wasn't the issue)
After a reset, initial position is sent to mesh about 1 minute after startup - after the first position is sent a flood of position updates start. Have tried with smart position broadcast both enabled and disabled (same behaviour).

From what I can see in noproto output, the position packet is sent successfully, the positionModule tries to cancel sending the packet that was sent successfully.

### Relevant log output

```Shell
01:52:33 21 [Power] Battery: usbPower=1, isCharging=1, batMv=4198, batPct=100
01:52:43 31 [NodeInfoModule] Sending our nodeinfo to mesh (wantReplies=1)
01:52:43 31 [NodeInfoModule] sending owner !84889ff8/White Portable/WHP
01:52:43 31 [NodeInfoModule] Update DB node 0x84889ff8, rx_time=1662429163
01:52:43 31 [NodeInfoModule] handleReceived(LOCAL) (id=0x53721d5e Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=4 WANTRESP rxtime=1662429163 priority=10)
01:52:43 31 [NodeInfoModule] No modules interested in portnum=4, src=LOCAL
01:52:43 31 [NodeInfoModule] Add packet record (id=0x53721d5e Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=4 WANTRESP rxtime=1662429163 priority=10)
01:52:43 31 [NodeInfoModule] Should encrypt MQTT?: 1
01:52:43 31 [NodeInfoModule] Using AES256 key!
01:52:43 31 [NodeInfoModule] ESP32 crypt fr=84889ff8, num=53721d5e, numBytes=48!
01:52:43 31 [NodeInfoModule] enqueuing for send (id=0x53721d5e Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x2d encrypted rxtime=1662429163 priority=10)
01:52:43 31 [NodeInfoModule] txGood=1,rxGood=0,rxBad=0
01:52:43 31 [RadioIf] Starting low level send (id=0x53721d5e Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x2d encrypted rxtime=1662429163 priority=10)
01:52:43 31 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=64, time 1132 ms
01:52:43 31 [RadioIf] AirTime - Packet transmitted : 1132ms
01:52:44 32 [RadioIf] Completed sending (id=0x53721d5e Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x2d encrypted rxtime=1662429163 priority=10)
01:52:54 41 [Power] Battery: usbPower=1, isCharging=1, batMv=4198, batPct=100
01:53:13 61 [PositionModule] Sending pos@6316a7cf:6 to mesh (wantReplies=1)
01:53:13 61 [PositionModule] Providing time to mesh 1662429193
01:53:13 61 [PositionModule] Position reply: time=1662429193, latI=XXX, lonI=-XXX
01:53:13 61 [PositionModule] Update DB node 0x84889ff8, rx_time=1662429193
01:53:13 61 [PositionModule] handleReceived(LOCAL) (id=0x53721d5f Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=3 WANTRESP rxtime=1662429193 priority=10)
01:53:13 61 [PositionModule] No modules interested in portnum=3, src=LOCAL
01:53:13 61 [PositionModule] Add packet record (id=0x53721d5f Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=3 WANTRESP rxtime=1662429193 priority=10)
01:53:13 61 [PositionModule] Should encrypt MQTT?: 1
01:53:13 61 [PositionModule] Using AES256 key!
01:53:13 61 [PositionModule] ESP32 crypt fr=84889ff8, num=53721d5f, numBytes=21!
01:53:13 61 [PositionModule] enqueuing for send (id=0x53721d5f Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x2d encrypted rxtime=1662429193 priority=10)
01:53:13 61 [PositionModule] txGood=2,rxGood=0,rxBad=0
01:53:13 61 [PositionModule] Using channel 0 (hash 0x2d)
01:53:13 61 [PositionModule] Using AES256 key!
01:53:13 61 [PositionModule] ESP32 crypt fr=84889ff8, num=53721d5f, numBytes=21!
01:53:13 61 [PositionModule] decoded message (id=0x53721d5f Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=3 WANTRESP rxtime=1662429193 priority=10)
01:53:13 61 [RadioIf] Starting low level send (id=0x53721d5f Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x2d encrypted rxtime=1662429193 priority=10)
01:53:13 61 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=37, time 804 ms
01:53:13 61 [RadioIf] AirTime - Packet transmitted : 804ms
01:53:14 61 [Power] Battery: usbPower=1, isCharging=1, batMv=4198, batPct=100
01:53:14 62 [RadioIf] Completed sending (id=0x53721d5f Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x2d encrypted rxtime=1662429193 priority=10)
01:53:18 66 [PositionModule] Sending pos@6316a7cf:6 to mesh (wantReplies=0)
01:53:18 66 [PositionModule] cancelSending id=0x53721d5f, removed=0
01:53:18 66 [PositionModule] Providing time to mesh 1662429198
01:53:18 66 [PositionModule] Position reply: time=1662429198, latI=XXX, lonI=-XXX
01:53:18 66 [PositionModule] Update DB node 0x84889ff8, rx_time=1662429198
01:53:18 66 [PositionModule] handleReceived(LOCAL) (id=0x53721d60 Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=3 rxtime=1662429198 priority=10)
01:53:18 66 [PositionModule] No modules interested in portnum=3, src=LOCAL
01:53:18 66 [PositionModule] Add packet record (id=0x53721d60 Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=3 rxtime=1662429198 priority=10)
01:53:18 66 [PositionModule] Should encrypt MQTT?: 1
01:53:18 66 [PositionModule] Using AES256 key!
[... 51 lines cut ...]
01:53:29 76 [RadioIf] AirTime - Packet transmitted : 780ms
01:53:29 77 [RadioIf] Completed sending (id=0x53721d62 Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x2d encrypted rxtime=1662429208 priority=10)
01:53:33 81 [PositionModule] Sending pos@6316a7cf:6 to mesh (wantReplies=0)
01:53:33 81 [PositionModule] cancelSending id=0x53721d62, removed=0
01:53:33 81 [PositionModule] Providing time to mesh 1662429213
01:53:33 81 [PositionModule] Position reply: time=1662429213, latI=XXX, lonI=-XXX
01:53:33 81 [PositionModule] Update DB node 0x84889ff8, rx_time=1662429213
01:53:33 81 [PositionModule] handleReceived(LOCAL) (id=0x53721d63 Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=3 rxtime=1662429213 priority=10)
01:53:33 81 [PositionModule] No modules interested in portnum=3, src=LOCAL
01:53:33 81 [PositionModule] Add packet record (id=0x53721d63 Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=3 rxtime=1662429213 priority=10)
01:53:33 81 [PositionModule] Should encrypt MQTT?: 1
01:53:33 81 [PositionModule] Using AES256 key!
01:53:33 81 [PositionModule] ESP32 crypt fr=84889ff8, num=53721d63, numBytes=19!
01:53:33 81 [PositionModule] enqueuing for send (id=0x53721d63 Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x2d encrypted rxtime=1662429213 priority=10)
01:53:33 81 [PositionModule] txGood=6,rxGood=0,rxBad=0
01:53:33 81 [PositionModule] Using channel 0 (hash 0x2d)
01:53:33 81 [PositionModule] Using AES256 key!
01:53:33 81 [PositionModule] ESP32 crypt fr=84889ff8, num=53721d63, numBytes=19!
01:53:33 81 [PositionModule] decoded message (id=0x53721d63 Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=3 rxtime=1662429213 priority=10)
01:53:34 81 [RadioIf] Starting low level send (id=0x53721d63 Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x2d encrypted rxtime=1662429213 priority=10)
01:53:34 81 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=35, time 780 ms
01:53:34 81 [RadioIf] AirTime - Packet transmitted : 780ms
01:53:34 81 [Power] Battery: usbPower=1, isCharging=1, batMv=4198, batPct=100
01:53:34 82 [RadioIf] Completed sending (id=0x53721d63 Fr0xf8 To0xff, WantAck0, HopLim3 Ch0x2d encrypted rxtime=1662429213 priority=10)
```

garthvh, 2022-09-06:
These values are working on the right intervals in the iOS app, what values did you try on the CLI?

	case fifteenSeconds = 15
	case thirtySeconds = 30
	case oneMinute = 60
	case fiveMinutes = 300
	case tenMinutes = 600
	case fifteenMinutes = 0
	case thirtyMinutes = 1800
	case oneHour = 3600
	case sixHours = 21600
	case twelveHours = 43200
	case twentyFourHours = 86400

thebentern, 2022-09-06:
@rohanki can you try it in one of the latest builds from master? We fixed some related issues with regard to position_broadcast_secs.

rohanki, 2022-09-06:
> These values are working on the right intervals in the iOS app, what values did you try on the CLI?
>
> ```
> case fifteenSeconds = 15
> case thirtySeconds = 30
> case oneMinute = 60
> case fiveMinutes = 300 [...]

I have tried 1 minute, 5 minutes and 30 seconds from the app. They all do the same. The setting is applied correctly on the device (ie: selecting one minute updates positionBroadcastSecs to 60), but the same runaway behaviour is shown whether set through the app or python cli..

Build 1.3.40.d8219fe is working perfectly.. At least with 60 seconds I'm seeing regular position broadcasts every 60 seconds :) Thankyou!

closed 2022-09-06 by rohanki