meshtastic/firmware #1679: [Bug]: 1.3.40.e87ecc2 beta 1 crashes with backtrace

Noki, 2022-09-07:
### Category

Other

### Hardware

T-Beam

### Firmware Version

1.3.40.e87ecc2

### Description

After doing a fresh install and setting the basics the node keeps crashing and rebooting with the backtrace as seen below.

### Relevant log output

```Shell
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
▒▒H??:??:?? 0

//\ E S H T /\ S T / C

??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=reset
??:??:?? 0 Filesystem files:
??:??:?? 0   /prefs/channels.proto (59 Bytes)
??:??:?? 0   /prefs/config.proto (121 Bytes)
??:??:?? 0   /prefs/db.proto (152 Bytes)
??:??:?? 0   /prefs/module.proto (18 Bytes)
??:??:?? 0 I2C device found at address 0x34
??:??:?? 0 axp192 PMU found
??:??:?? 0 I2C device found at address 0x3c
??:??:?? 0 0x7 subtype probed in 2 tries
??:??:?? 0 ssd1306 display found
??:??:?? 0 2 I2C devices found
??:??:?? 0 Meshtastic hwvendor=4, swver=1.3.40.e87ecc2-d
??:??:?? 0 Setting random seed 3511673233
??:??:?? 0 Total heap: 239372
??:??:?? 0 Free heap: 207236
??:??:?? 0 Total PSRAM: 4194252
??:??:?? 0 Free PSRAM: 4194252
??:??:?? 0 NVS: UsedEntries 238, FreeEntries 392, AllEntries 630, NameSpaces 8
??:??:?? 0 Setup Preferences in Flash Storage
??:??:?? 0 Number of Device Reboots: 61
??:??:?? 0 Initializing NodeDB
??:??:?? 0 Loading /prefs/db.proto
??:??:?? 0 Loaded saved devicestate version 16
??:??:?? 0 Loading /prefs/config.proto
??:??:?? 0 Loaded saved config version 16
??:??:?? 0 Loading /prefs/module.proto
??:??:?? 0 Loaded saved moduleConfig version 16
??:??:?? 0 Loading /prefs/channels.proto
??:??:?? 0 Loaded saved channelFile version 16
??:??:?? 0 Number of Device Reboots: 61
??:??:?? 0 Expanding short PSK #1
??:??:?? 0 Wanted region 3, using EU868
??:??:?? 0 region=3, NODENUM=0xbdf0c808, dbsize=1
[... 159 lines cut ...]
??:??:?? 31 [NodeInfoModule] Should encrypt MQTT?: 1
??:??:?? 31 [NodeInfoModule] Expanding short PSK #1
??:??:?? 31 [NodeInfoModule] Using AES128 key!
??:??:?? 31 [NodeInfoModule] ESP32 crypt fr=bdf0c808, num=2a0d6260, numBytes=40!
??:??:?? 31 [NodeInfoModule] publish msh/1/c/LongF/!bdf0c808, 83 bytes
??:??:?? 31 [NodeInfoModule] enqueuing for send (id=0x2a0d6260 Fr0x08 To0xff, WantAck0, HopLim3 Ch0x6e encrypted priority=10)
??:??:?? 31 [NodeInfoModule] txGood=1,rxGood=0,rxBad=0
??:??:?? 31 [RadioIf] Starting low level send (id=0x2a0d6260 Fr0x08 To0xff, WantAck0, HopLim3 Ch0x6e encrypted priority=10)
??:??:?? 31 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=56, time 1034 ms
??:??:?? 31 [RadioIf] AirTime - Packet transmitted : 1034ms
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40001277  PS      : 0x00060030  A0      : 0x800f2c95  A1      : 0x3ffd5600
A2      : 0x00000000  A3      : 0x3ffc419c  A4      : 0x00000000  A5      : 0x00002c40
A6      : 0x00000000  A7      : 0x3ffceaac  A8      : 0x80125de1  A9      : 0x3ffd5630
A10     : 0x00000001  A11     : 0x0000006e  A12     : 0x00000001  A13     : 0x00000038
A14     : 0x0000006d  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x40001277:0x3ffd5600 0x400f2c92:0x3ffd5610 0x400f2d2d:0x3ffd56d0 0x401f3287:0x3ffd56f0 0x400fe415:0x3ffd5710 0x400f28f5:0x3ffd5750 0x400d572e:0x3ffd5770 0x400f79b5:0x3ffd5790 0x400dd9a0:0x3ffd57c0 0x4012cc65:0x3ffd57e0

Rebooting...
```

thebentern, 2022-09-07:
Looks like it's choking on some mqtt messages. Not entirely sure yet, but I wonder if it could be older versioned message protobufs coming through the public MQTT server.

Here's the decoded trace:
```
    0x400f2c92: MQTT::onPublish(char*, unsigned char*, unsigned int) at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mqtt/MQTT.cpp:70
    0x400f2d2d: MQTT::mqttCallback(char*, unsigned char*, unsigned int) at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mqtt/MQTT.cpp:23
    0x401f3287: std::_Function_handler::_M_invoke(std::_Any_data const&, char*&&, unsigned char*&&, unsigned int&&) at /home/runner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
    0x400fe415: std::function::operator()(char*, unsigned char*, unsigned int) const at /home/runner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:2271
    0x400fe415: PubSubClient::loop() at /home/runner/work/Meshtastic-device/Meshtastic-device/.pio/libdeps/tbeam/PubSubClient/src/PubSubClient.cpp:416
    0x400f28f5: MQTT::runOnce() at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mqtt/MQTT.cpp:199
    0x400d572e: concurrency::OSThread::run() at /home/runner/work/Meshtastic-device/Meshtastic-device/src/concurrency/OSThread.cpp:78
    0x400f79b5: ThreadController::runOrDelay() at /home/runner/work/Meshtastic-device/Meshtastic-device/.pio/libdeps/tbeam/Thread/ThreadController.cpp:59
    0x400dd9a0: loop() at /home/runner/work/Meshtastic-device/Meshtastic-device/src/main.cpp:481
    0x4012cc65: loopTask(void*) at /home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19

```

garthvh, 2022-09-07:
This is not happening with our 1.3 only dev MQTT broker

Noki, 2022-09-12:
> This is not happening with our 1.3 only dev MQTT broker

Just use the public MQTT broker, it is constantly happening there. ;-)

andrekir, 2022-09-19:
this should be fixed with 1.3.41.

Noki, 2022-09-21:
Still present in 1.3.41:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40001277  PS      : 0x00060a30  A0      : 0x800f5c99  A1      : 0x3ffd5550
A2      : 0x00000000  A3      : 0x3ffc418c  A4      : 0x00000000  A5      : 0x0000c180
A6      : 0x00000000  A7      : 0x3ffd37c0  A8      : 0x80128735  A9      : 0x3ffd5580
A10     : 0x00000001  A11     : 0x00000074  A12     : 0x00000001  A13     : 0x00000038
A14     : 0x00000073  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x40001277:0x3ffd5550 0x400f5c96:0x3ffd5560 0x400f5d31:0x3ffd5620 0x401f5dab:0x3ffd5640 0x401014d5:0x3ffd5660 0x400f58f9:0x3ffd56a0 0x400d832a:0x3ffd56c0 0x400fa9f9:0x3ffd56e0 0x400e08a4:0x3ffd5710 0x4012f5b9:0x3ffd5730

Rebooting...
```

caveman99, 2022-10-05:
please test again with latest 1.3.42-pre firmware from github...

closed 2022-10-05 by caveman99

garthvh, 2022-10-13:
This has now been released in 1.3.43, the public MQTT should be able to handle both 1.2 and 1.3