meshtastic/firmware #1680: Set Fixed Position changed back to GPS Postion after some time[Bug]:

Bergstern, 2022-09-08:
### Category

Other

### Hardware

Rak4631

### Firmware Version

1.3.40.e87ecc2-d

### Description

When set a fixed position with the command "meshtastic --setlat 50 --setlon 50" the cli confirmed the changes and at the meshtastic info there are the right coords. After a while the coords. jumped back to the gps coordinates of the current position of the node. "hasGps": is at true, should be at false.

garthvh, 2022-09-11:
hasGPS indicates that there is a hardware GPS, disabling it does not flip this boolean

garthvh, 2022-09-11:
You also need to disable GPS for this to work, as "meshtastic --setlat 50 --setlon 50" just adds a regular position and fixed position uses the last set position, if gps is on that position will be updated based on the GPS settings.

Bergstern, 2022-09-11:
[image]

Bergstern, 2022-09-11:
Thats displaed at the debug from the android software. Gps already is disabled. Thats reason i think its a bug. If i set gps at disabled and set a fixed position the gps should turn off?

garthvh, 2022-09-11:
yeah seems like a bug, I think this feature is mostly used to add location when a device has a gps, but it seems like it should use the fixed location and turn off the gps if the gps is disabled and fixed position is set. @thebentern seem right?

caveman99, 2022-10-04:
GPS won't turn off until after next reboot if it is set to false. @Bergstern can you double check these settings, then reboot the device and set a fixed position? that should not change any more.

Bergstern, 2022-10-09:
@caveman99 i cant test it for now because i dismount the gps module. And the rak isnt at home at the moment. Sry

caveman99, 2022-10-12:
closing for now cause i can't reproduce. Please reopen if still a problem.

closed 2022-10-12 by caveman99

IZ1IVA, 2022-10-14:
Setting a fixed position is still not reliable on 1.3.44 on tlora-v2-1-1.6 and not possibile at all on tbeam.