meshtastic/firmware #1681: Battery voltage and charge level show up only sporadic

Bergstern, 2022-09-08:
When i connect the solar panel to the solar input of the RAK19007 the battery voltage and charge level show up only sporadic and the "plug" icon is displayed. when i connect with usb the battery voltage and charge level in percent shows up continuous. At discord they tell me its because technical limitations that the module cant differentiate between solar and main power source, but i think in this way it make no sense too. 
would it be possible to always show up voltage and charge level?
Thank u
Bg Andre

andrekir, 2022-09-08:
devices generate `battery_level` data. apps just use the received value. 0 means charging/USB powered (ie. power plug), 1-100 charge %.

Bergstern, 2022-09-09:
Thank u. So i close this issue

closed 2022-09-09 by Bergstern