meshtastic/firmware #1707: [Bug]: 1.3 Serial Module Does not work

garthvh, 2022-09-21:
### Category

Serial

### Hardware

T-Beam

### Firmware Version

1.3.41

### Description

When using a tbeam on a Mac, the serial module is not working

Setup:

Tbeam 1.1 with: 

Serial Enabled
Echo Enabled
RXD set to pin 35
TX set to pin 15
Baud default
Mode: Text message

Expected behavior 
Using tio for serial

echo 'test serial messge' | tio /dev/tty.wchusbserial1130 --response-wait    would send a message that will be received on the broadcast channel

geeksville, 2022-09-21:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/lilygo-ttgo-1-3-40-915mhz-lora32-hardware-serial-not-usb/6487/6

garthvh, 2022-09-21:
[image]

geeksville, 2022-09-29:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/help-with-serial-module-transmit-receive-text-message-on-tbeam/6527/4

garthvh, 2022-10-04:
This is what I get in the serial log when sending data

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[???mum????2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

garthvh, 2022-10-04:
And 
??:??:?? 2 [SerialModule] Initializing serial peripheral interface
[  2646][E][HardwareSerial.cpp:506] setRxBufferSize(): RX Buffer can't be resized when Serial is already running.

caveman99, 2022-10-05:
the RX buffer compaint comes from using the new ESP32 toolkit. I'll have a look

caveman99, 2022-10-05:
i checiked in something, can you retest with latest trunk?

garthvh, 2022-10-14:
Test configs

38400 Baud

For a tbeam the module config is 
serial.enabled true
serial.rxd 35
serial.txd 15
serial.timeout 1000
serial.echo true
serial.mode TEXTMSG

For rak use

serial.enabled  true
serial.rxd  15
serial.txd 16
serial.timeout  1000
serial.echo  true
serial.mode TEXTMSG

lesykm, 2022-10-15:
Hmm, is it OK that for RAK pins 15 and 16 are used for GPS?

https://github.com/meshtastic/Meshtastic-device/blob/ade32b1827b7dc554716625e50c3ccfd8b87085b/variants/rak4631/variant.h#L101-L102

https://github.com/meshtastic/Meshtastic-device/blob/ade32b1827b7dc554716625e50c3ccfd8b87085b/variants/rak4631/variant.h#L198-L199

lesykm, 2022-10-15:
After I fixed pins according to original variant.h: https://github.com/RAKWireless/RAK-nRF52-Arduino/blob/ca1dd0b2b17a9bd12a6d9a4d0f39b528e2104efc/variants/WisCore_RAK4631_Board/variant.h#L122-L123

Serial module started working as desired.

So in this repository variant.h is incorrect

lesykm, 2022-10-15:
Yes, just re-checked, having this only change:

```
-#define PIN_SERIAL2_RX (8)
-#define PIN_SERIAL2_TX (6)
+#define PIN_SERIAL2_RX (19)
+#define PIN_SERIAL2_TX (20)
```

Serial module fully works.

garthvh, 2022-10-15:
Use the module config, manually updating the firmware is not the desired result

thebentern, 2022-10-15:
Yes `meshtastic --set serial.rxd 15 --set serial.txd 16` should do it

lesykm, 2022-10-16:
Right, I just wanted to mention that serial module works with 19/20 pins :)

garthvh, 2022-10-16:
@jfirwin we have a bunch of serial module stuff working that we should document.  Anyone here with a working config it would be great if we could get a board picture and your config.

closed 2022-10-17 by thebentern

lesykm, 2022-10-17:
Just small note - text messages sent through serial appeared in main channel and not in "serial" channel. If it is desired, then OK.

thebentern, 2022-10-17:
That's a good thing. If they're sent over the serial portnum, they wouldn't show up in the messages for the apps. Which is a use case for some.