meshtastic/firmware #1708: Feature Request/Proposal: Implement onboard firmware/flash protection

lesykm, 2022-09-21:
Right now it is possible to dump flash from ESP32/nRF52 chips and extract any information from the flash.
Proposal: Implement flash encryption as provided in documentation:

ESP32 links:
1. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html
2. https://blog.espressif.com/understanding-esp32s-security-features-14483e465724
3. https://circuitdigest.com/article/how-to-secure-esp32-firmware-and-flash-memory-using-esp-idf-framework

nRF52 links
1. https://devzone.nordicsemi.com/f/nordic-q-a/23972/cybersecurity-features-for-nrf52-chips
2. https://devzone.nordicsemi.com/f/nordic-q-a/25926/flash-protection-nrf52840-sdk-14-0-0

caveman99, 2022-09-21:
The thing is - If you get physical access to a device, there are easier ways to get to the important bits than dumping the flash. You can leave a device outside without the encryption keys (albeit on the same rado settings) and it will relay your messages anyway. So this encryption, while technically doable, would not add anytzhing to device security.

lesykm, 2022-09-21:
Sure, this can be (less priority) part of the project how to hardened autonomous node which was stolen from compromising entire network (i.e. disposable nodes). In certificates world this is done using certificate revocation: https://en.wikipedia.org/wiki/Certificate_revocation_list
May be something similar can be implemented in Meshtastic? (sure not in 1.3 and out of scope of this issue which is very specific).

Out of curiosity can you please point me/enumerate these easier ways to get keys/other important info from inside the node?
Thanks!

lesykm, 2022-09-21:
I currently play with just ESP32 by flashing firmware into it and learning parameters but real hardware are on the way. Regarding "connect over serial" - I thought `meshtastic --set device.serial_disabled true` should disable it ( according to https://github.com/meshtastic/Meshtastic-device/issues/638) but it did not in my case 0_o - just checked - `meshtastic --info` still gives me information, is it a bug or expected?

lesykm, 2022-09-21:
Filed https://github.com/meshtastic/Meshtastic-device/issues/1713 just in case.

closed 2023-05-23 by thebentern