meshtastic/firmware #1709: Feature Request/Proposal: Provide ability to disable (through admin channel) pairing new BT devices

lesykm, 2022-09-21:
Right now anyone who will find a node can bind their phone/other device to it and get access to the network.
It would be nice to add ability to completely turn off (and later optionally turn on) pairing of new devices to given nodes via admin channel commands.

garthvh, 2022-09-21:
You can disable Bluetooth already meshtastic --set bluetooth.enabled false

lesykm, 2022-09-21:
As I mentioned just disabling bluetooth will disable usage of the end nodes. Just disable pairing while enable bluetooth running will be better.
Regarding fixed pin - it will be visible on a screen, right?

garthvh, 2022-09-23:
The fixed pin works without a device screen fine, it defaults to 123456 and you can set it from the CLI.  I am not aware of any way in the nimbleBLE stack on device or the various central managers used in client apps to have a device advertise but only to certain other devices.

lesykm, 2022-09-27:
Few links to read:
* https://mynewt.apache.org/latest/network/ble_sec.html - Pairing should be disabled

Potential BLE framework (NimBLE) solutions:
* User whitelist (add paired device to whitelist): https://github.com/apache/mynewt-nimble/blob/f7ebfaf95b44013c8ab938edbab2bf98bebc7d8b/nimble/syscfg.yml#L42
* Disable pairing using following parameters: https://github.com/apache/mynewt-nimble/blob/46df736023e2fa76c425418073d80ff1d76d54ae/nimble/host/syscfg.yml#L97-L102

Potential Meshtastic-only solution:
Enable bluetooth "hardened: mode:
* Disabe "DisablePIN" mode: https://github.com/meshtastic/Meshtastic-device/blob/f7b12f06956d6ed4d1410d7ec36c83398946749e/src/nimble/NimbleBluetooth.cpp#L206-L225
* Disable default static pin: https://github.com/meshtastic/Meshtastic-device/blob/f7b12f06956d6ed4d1410d7ec36c83398946749e/src/nimble/NimbleBluetooth.cpp#L75-L78
* Disable passkey showing: https://github.com/meshtastic/Meshtastic-device/blob/f7b12f06956d6ed4d1410d7ec36c83398946749e/src/nimble/NimbleBluetooth.cpp#L88

caveman99, 2022-09-27:
The whitelist looks like a good approach on that. I can see the reasoning behind wanting to use bluetooth but disabling signing up new devices.
Bluetooth (or better BLE) in its current state was a major source of problems for a long time, that's why some things are like they are right now. It has improved a lot since we used newer versions of the BLE stacks on the device., so we might start enabling security features again.