meshtastic/firmware #1713: [Bug]: setting device.serial_enabled to false does not disable serial console.

lesykm, 2022-09-21:
### Category

Serial

### Hardware

T-Lora v1

### Firmware Version

1.3.40.e87ecc2-d

### Description

1. `meshtastic --set device.serial_enabled false`
2. `meshtastic --info | grep serial
Preferences: { "device": { "serialDisabled": true, "debugLogEnabled": true, "ntpServer": "0.pool.ntp.org" }, "position": { "positionFlags": 3 }, "power": { "lsSecs": 300 }, "wifi": {}, "display": {}, "lora": {}, "bluetooth": { "enabled": true, "mode": "NoPin", "fixedPin": 123456 } }
Module preferences: { "mqtt": {}, "serial": {}, "externalNotification": {}, "rangeTest": {}, "telemetry": {}, "cannedMessage": {} }`

To my understanding  "serialDisabled": true should make it impossible to use console

lesykm, 2022-09-21:
Interestingly: 
https://github.com/meshtastic/Meshtastic-device/search?q=serial_disabled is empty instead we have:
https://github.com/meshtastic/Meshtastic-device/search?q=serial_enabled
However:
```
meshtastic --set device.serial_enabled false
Connected to radio
LocalConfig and LocalModuleConfig do not have attribute device.serial_enabled.
```
Something is off here
...

caveman99, 2022-09-24:
you have to update your CLI. needs 1.3.35 - you can do that with `pip3 install --upgrade meshtastic --pre`

lesykm, 2022-09-24:
@caveman99 Still does not work:

```
meshtastic --version
1.3.35
```

```
meshtastic --info
Connected to radio

Owner: Meshtastic ea54 (ea54)
My info: { "myNodeNum": 2747853396, "maxChannels": 8, "firmwareVersion": "1.3.42.2f74f9ca", "errorCode": "NO_RADIO", "errorAddress": 421, "errorCount": 3, "rebootCount": 16, "messageTimeoutMsec": 300000, "minAppVersion": 20300, "hasWifi": true }
Nodes in mesh:  {'num': 2747853396, 'user': {'id': '!a3c8ea54', 'longName': 'Meshtastic ea54', 'shortName': 'ea54', 'macaddr': 'c8:c9:a3:c8:ea:54', 'hwModel': 'TLORA_V2_1_1P6'}, 'position': {}, 'deviceMetrics': {}}

Preferences: { "device": {}, "position": { "positionBroadcastSecs": 900, "positionBroadcastSmartEnabled": true, "gpsEnabled": true, "gpsUpdateInterval": 120, "gpsAttemptTime": 900, "positionFlags": 3 }, "power": { "waitBluetoothSecs": 60, "meshSdsTimeoutSecs": 7200, "sdsSecs": 4294967295, "lsSecs": 300, "minWakeSecs": 10 }, "network": { "ntpServer": "0.pool.ntp.org" }, "display": { "screenOnSecs": 600 }, "lora": { "hopLimit": 3, "txEnabled": true, "txPower": 30 }, "bluetooth": { "enabled": true, "mode": "FIXED_PIN", "fixedPin": 123456 } }

Module preferences: { "mqtt": {}, "serial": {}, "externalNotification": {}, "rangeTest": {}, "telemetry": { "deviceUpdateInterval": 900, "environmentUpdateInterval": 900 }, "cannedMessage": {} }

Channels:
  PRIMARY psk=default { "psk": "AQ==" }

Primary channel URL: https://www.meshtastic.org/e/#CgMSAQESBkADSAFQHg
```

```
meshtastic --set device.serial_enabled false
Connected to radio
Set device.serial_enabled to false
Writing modified preferences to device
```

```
Connected to radio

Owner: Meshtastic ea54 (ea54)
My info: { "myNodeNum": 2747853396, "maxChannels": 8, "firmwareVersion": "1.3.42.2f74f9ca", "errorCode": "NO_RADIO", "errorAddress": 421, "errorCount": 4, "rebootCount": 17, "messageTimeoutMsec": 300000, "minAppVersion": 20300, "hasWifi": true }
Nodes in mesh:  {'num': 2747853396, 'user': {'id': '!a3c8ea54', 'longName': 'Meshtastic ea54', 'shortName': 'ea54', 'macaddr': 'c8:c9:a3:c8:ea:54', 'hwModel': 'TLORA_V2_1_1P6'}, 'position': {}, 'deviceMetrics': {}}

Preferences: { "device": {}, "position": { "positionBroadcastSecs": 900, "positionBroadcastSmartEnabled": true, "gpsEnabled": true, "gpsUpdateInterval": 120, "gpsAttemptTime": 900, "positionFlags": 3 }, "power": { "waitBluetoothSecs": 60, "meshSdsTimeoutSecs": 7200, "sdsSecs": 4294967295, "lsSecs": 300, "minWakeSecs": 10 }, "network": { "ntpServer": "0.pool.ntp.org" }, "display": { "screenOnSecs": 600 }, "lora": { "hopLimit": 3, "txEnabled": true, "txPower": 30 }, "bluetooth": { "enabled": true, "mode": "FIXED_PIN", "fixedPin": 123456 } }

Module preferences: { "mqtt": {}, "serial": {}, "externalNotification": {}, "rangeTest": {}, "telemetry": { "deviceUpdateInterval": 900, "environmentUpdateInterval": 900 }, "cannedMessage": {} }

Channels:
  PRIMARY psk=default { "psk": "AQ==" }

Primary channel URL: https://www.meshtastic.org/e/#CgMSAQESBkADSAFQHg
```

hard_reboot_just_in_case

```
Connected to radio

Owner: Meshtastic ea54 (ea54)
My info: { "myNodeNum": 2747853396, "maxChannels": 8, "firmwareVersion": "1.3.42.2f74f9ca", "errorCode": "NO_RADIO", "errorAddress": 421, "errorCount": 4, "rebootCount": 25, "messageTimeoutMsec": 300000, "minAppVersion": 20300, "hasWifi": true }
Nodes in mesh:  {'num': 2747853396, 'user': {'id': '!a3c8ea54', 'longName': 'Meshtastic ea54', 'shortName': 'ea54', 'macaddr': 'c8:c9:a3:c8:ea:54', 'hwModel': 'TLORA_V2_1_1P6'}, 'position': {}, 'deviceMetrics': {}}

Preferences: { "device": {}, "position": { "positionBroadcastSecs": 900, "positionBroadcastSmartEnabled": true, "gpsEnabled": true, "gpsUpdateInterval": 120, "gpsAttemptTime": 900, "positionFlags": 3 }, "power": { "waitBluetoothSecs": 60, "meshSdsTimeoutSecs": 7200, "sdsSecs": 4294967295, "lsSecs": 300, "minWakeSecs": 10 }, "network": { "ntpServer": "0.pool.ntp.org" }, "display": { "screenOnSecs": 600 }, "lora": { "hopLimit": 3, "txEnabled": true, "txPower": 30 }, "bluetooth": { "enabled": true, "mode": "FIXED_PIN", "fixedPin": 123456 } }

Module preferences: { "mqtt": {}, "serial": {}, "externalNotification": {}, "rangeTest": {}, "telemetry": { "deviceUpdateInterval": 900, "environmentUpdateInterval": 900 }, "cannedMessage": {} }

Channels:
  PRIMARY psk=default { "psk": "AQ==" }

Primary channel URL: https://www.meshtastic.org/e/#CgMSAQESBkADSAFQHg
```

Am I doing something wrong?

Also, just in case:

```
meshtastic --set device.serial_disabled true
Connected to radio
LocalConfig and LocalModuleConfig do not have attribute device.serial_disabled.
```

caveman99, 2022-09-24:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;} I'll have a look tomorrow, looks like it's not even saved. Anyway the value is only read during device startup, so you should See it after the reset but it's only active after a reboot. We can't de-init the console as of now.                                                                                                                                             Gesendet über BlackBerry Hub+ Posteingang für Android

garthvh, 2022-09-24:
This has been modified to meshtastic --set device.serial_enabled = false

garthvh, 2022-09-24:
Device also reboots automatically after saving now as well in firmwares 1.3.41+

lesykm, 2022-09-25:
> #response_container_BBPPID{font-family: initial; font-size:initial; color: initial;} I'll have a look tomorrow, looks like it's not even saved. Anyway the value is only read during device startup, [...]

I did reset the device, including detaching of power.

lesykm, 2022-09-25:
> This has been modified to meshtastic --set device.serial_enabled = false

I did execute this command, please see my listings.

lesykm, 2022-09-25:
> Device also reboots automatically after saving now as well in firmwares 1.3.41+

I did all kind of reboots I can imagine just to be sure :)

lesykm, 2022-09-27:
@caveman99 this issue is reproducible on many devices - can it be re-opened?

caveman99, 2022-09-27:
we've had a bit of erratic behaviour with device filesystem. I'll revisit this after we get CI running again. Looks like an issue where the setting is not saved to the protobuf.

caveman99, 2022-09-27:
Just a note to me: Serial API and Serial Console (e.g. debug logging) are two different codepaths. Maybe this setting switches debug log off but keeps the api adapter active. OTOH when i checked it effectively does not open the serial port ... so no API action either.

caveman99, 2022-09-27:
This is a tricky one. Here's the code in question:

    if (!config.has_device || config.device.serial_enabled) {
        consoleInit(); // Set serial baud rate and init our mesh console
    }

On a new flashed device the entire device structure is empty/default. and setting serial_enabled to 0 will effectively not save that info since it's also assumed to be the default. The way this is crafted now, you have to set another non-default value in device to actually disable the console.

@thebentern @sachaw maybe thgis is another case where we have to enforce the default of 1/enabled so we can change that if statement.

sachaw, 2022-09-27:
This is already true by default, implemented in https://github.com/meshtastic/Meshtastic-device/commit/60b4dbfdcd4d85355295756c0e3330b556020b61 or am I missing something here?

garthvh, 2022-10-04:
It seems to accept the command but does not save the value

When I do  --set device.serial_enabled false

and then meshtastic --info I get 

Preferences: { "device": {},

caveman99, 2022-10-04:
the 0/false value is not printed, but it is stored correctly. Problem is nodeDB is not loaded (yet) when we check for this. I will submit a proposed solution later on.

commit e5605cc6 (2022-10-05): Merge pull request #1753 from meshtastic/bug-1713

fix #1713
commit f7331a2e (2022-10-05): fix #1713

stop output to serial debug after config has been loaded (if serial is disabled), and only accept protobuf packets from that point on.
closed 2022-10-05 by caveman99