meshtastic/firmware #1714: iOS Nodes do not reflect new names after change

ghostop14, 2022-09-21:
Testing 1.3. 40 and 41, If a device's owner name is changed, the node list appears to keep the old name rather than reflecting the change.

garthvh, 2022-09-21:
There is an item here to reboot the node after a name change so the changes can properly propagate https://github.com/meshtastic/Meshtastic-device/issues/1590

garthvh, 2022-09-21:
The other option would be to send a node info packet automaticlly after the change, I need verification that the change took place before the data in iOS will be updated.

ghostop14, 2022-09-21:
If the name/data is stored on the iOS side, even the ability for the user to manually delete it from the list and let it repopulate when the device sends out an info packet next might be a quick solution?

garthvh, 2022-09-21:
I really avoid cleaning up any data in the app, the app is just a view into the data from the mesh and any cleanup I have done causes a lot of user confusion, I think we are going to add a reboot.

closed 2022-09-22 by thebentern