meshtastic/firmware #1730: [Bug]: meshtastic --set device.factory_reset no longer works

simplyplant, 2022-09-28:
### Category

Other

### Hardware

T-Beam

### Firmware Version

v1.3.41

### Description

In v1.3.40  the command  'meshtastic --set device.factory_reset true' worked fine.

In v1.3.41 it doesn't anymore.

Output in v1.3.41
...
LocalConfig and LocalModuleConfig do not have attribute device.factory_reset.
....

I have already updated meshtastic --  pip3 install --upgrade meshtastic --pre

I do find this command rather useful for my project

Many thanks for your excellent work.

garthvh, 2022-09-28:
It is not in the device config anymore just meshtastic --set factory_reset true

closed 2022-09-28 by garthvh

thebentern, 2022-09-28:
Try meshtastic --factory-reset

simplyplant, 2022-09-28:
thebentern

Thank you. Thank you Thank you. This is what I was after.

I have a whole lot of these boards so this command helps a lot and saves a lot of time.

Very best regards and thanks for your contribution.